openwrt/luci #555: Time Synchronization block shows up in all 3 tabs of system/system

Bering, 2015-11-22:
Hi,

When going to the **System** dropdown and selecting the **system** page, there are 3 tabs but all 3 show the Time Synchronization block.

I suggest moving the Time Synchronization block in the first tab?

hnyman, 2015-11-22:
Hmm. 
to be exact, you mean that the time synchronization block is shown in all three tabs in addition to the otherwise correct content (general, logging, style). At least that is, how it works for me.

(seems to be the same on both trunk and CC15.05)

I guess that it is unnecessary to have it displayed on all the tabs.

Bering, 2015-11-22:
Yep, exactly. After revisiting all the other pages though, I think that it
might actually be as per design. I found it a bit confusing so I guess
newbies do to :-)

hnyman, 2015-11-22:
I feel that it should actually be either only on the timezone selection page or on a new fourth tab.

And yes, there are some other pages where similar behaviour is seen.

jow-, 2015-11-22:
Due to the structure of the cbi framework this is hard to fix, adding a lot of new untested code for very little gain.

hnyman, 2015-11-28:
closed as "wontfix"

closed 2015-11-28 by hnyman