openwrt/luci #556: luci-base: switch to luci-lib-jsonc #fbfa62627f66f98b6dd37ed6050c9b99f2479881 missing jsonc.lua

HenryGiraldo, 2015-11-24:
Updating to the lastest LuCI, could not be found jsonc.lua
<pre>
/usr/bin/lua: /usr/lib/lua/luci/util.lua:12: module 'luci.jsonc' not found:
uhttpd[1746]:       no field package.preload['luci.jsonc']
uhttpd[1746]:       no file './luci/jsonc.lua'
uhttpd[1746]:       no file '/usr/share/lua/luci/jsonc.lua'
uhttpd[1746]:       no file '/usr/share/lua/luci/jsonc/init.lua'
uhttpd[1746]:       no file '/usr/lib/lua/luci/jsonc.lua'
uhttpd[1746]:       no file '/usr/lib/lua/luci/jsonc/init.lua'
uhttpd[1746]:       no file './luci/jsonc.so'
uhttpd[1746]:       no file '/usr/lib/lua/luci/jsonc.so'
uhttpd[1746]:       no file '/usr/lib/lua/loadall.so'
uhttpd[1746]:       no file './luci.so'
uhttpd[1746]:       no file '/usr/lib/lua/luci.so'
uhttpd[1746]:       no file '/usr/lib/lua/loadall.so'
</pre>

jow-, 2015-11-24:
How did you update?

HenryGiraldo, 2015-11-24:
I did a pull from master. 
I update the LuCI feed in openwrt, the jsonc dependency was added and clean/compile LuCI.
Should jsonc.lua file in the repository or it is autogenerated? I did a clean clone and I can't found it.

jow-, 2015-11-24:
There is no jsonc.lua, but jsonc.so which is provided by luci-lib-jsonc. Did you run "make defconfig" after updating your feeds to account for the new dependency?

HenryGiraldo, 2015-11-24:
Working now, sorry for the mistake.
For some strange reason the .config wasn't correctly saved.

closed 2015-11-24 by HenryGiraldo