openwrt/luci #557: Sanitization of hostnames in DHCP Static Leases regarding underscore

JamesT42, 2015-11-25:
Hello!
The underscore is not sanitized in the hostname boxes in DHCP/Static Leases which leads to a partially non-working DHCP server. It took me many hours to find this bug.
For example i named a host abc_xyz, it would then randomly after a few days not be able to get an IP adress assigned any more. After changing it to abc-xyz it's now working again.

hnyman, 2015-11-25:
Hmmm.
I thought that this might be fixed by adding a "hostname" datatype validation to the static lease code.
But there is already that validation:
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/dhcp.lua#L224
```
name.datatype = "hostname"
```

However, the problem is that the current "hostname" validation seems to allow _ :
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/cbi/datatypes.lua#L168
```
function hostname(val)
	if val and (#val < 254) and (
	   val:match("^[a-zA-Z_]+$") or
	   (val:match("^[a-zA-Z0-9_][a-zA-Z0-9_%-%.]*[a-zA-Z0-9]$") and
	    val:match("[^0-9%.]"))
	) then
```

The RFCs (952,1123) and Wikipedia etc. agree that hostnames are not allowed to contain _
https://tools.ietf.org/html/rfc952
https://tools.ietf.org/html/rfc1123
https://en.wikipedia.org/wiki/Hostname#Restrictions_on_valid_host_names

> The Internet standards (Requests for Comments) for protocols mandate that component hostname labels may contain only the ASCII letters 'a' through 'z' (in a case-insensitive manner), the digits '0' through '9', and the hyphen ('-'). The original specification of hostnames in RFC 952, mandated that labels could not start with a digit or with a hyphen, and must not end with a hyphen. However, a subsequent specification (RFC 1123) permitted hostname labels to start with digits. No other symbols, punctuation characters, or white space are permitted.
> 
> While a hostname may not contain other characters, such as the underscore character (_), other DNS names may contain the underscore.[4] 

The validation seems to have intentionally relaxed by this commit: 
https://github.com/openwrt/luci/commit/c802c941cb5892ac43ee865594d3e06b60c00ace

hnyman, 2015-11-28:
@jow- 
Any comment to disabling _ from hostname validation? 

Leave validation as it is? 
Change it to eliminate _ from hostname?

JamesT42, 2015-11-28:
Is it maybe possible to just display a warning if _ is used? It may work fine for some people.

cshoredaniel, 2016-12-09:
I think that hostname vs dnsname should be different datatypes per-RFCs hostname can't have underscore, but other parts of the DNS names can.  The hostname part of the DNS name needs to be handled separately from the domain name (which can contain underscores).  Also need to double check where hostname datatype is used and modify appropriately.  I'm thinking base types of hostname, dnsname, and a type that calls them called fqdn and hostnameorfqdn (domain names ought to be handled by 'dnsname').

systemcrash, 2024-05-01:
error checks prevent this in luci today

closed 2024-05-01 by systemcrash