openwrt/luci #568: OpenWrt firewall - problem code 500

lukaszk14, 2015-12-09:
500 Internal Server Error
Sorry, but the server encountered an unexpected error.
/usr/lib/lua/luci/dispatcher.lua:360: Failed to execute function dispatcher target for entry '/admin/network/firewall'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:360: Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/zones'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'cbi/firewall_zoneforwards'.
A runtime error occured: /usr/lib/lua/luci/model/firewall.lua:61: bad argument #2 to 'get' (string expected, got nil)
stack traceback:
    [C]: in function 'assert'
    /usr/lib/lua/luci/dispatcher.lua:360: in function 'dispatch'
    /usr/lib/lua/luci/dispatcher.lua:135: in function

jow-, 2015-12-09:
Attach your /etc/config/firewall please

jow-, 2015-12-09:
Also the output of `opkg list_installed | grep luci`

lukaszk14, 2015-12-09:
I found a problem. 

Duplicate type the:
config zone 
config zone

in /etc/config/firewall

I apologize for the problem.

jow-, 2015-12-17:
Allright.

closed 2015-12-17 by jow-