openwrt/luci #569: luci-app-olsr: error on neighbors table

FreifunkUFO, 2015-12-09:
Problem with access to the OLSR Neighbors-Page:

/usr/lib/lua/luci/dispatcher.lua:452: Failed to execute call dispatcher target for entry '/freifunk/olsr/neighbors'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/olsr.lua:175: attempt to index field 'iwdata' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

tested with openwrt-trunk:
root@113-76:~# uname -a
Linux 113-76 4.1.13 #9 Mon Dec 7 12:18:37 CET 2015 mips GNU/Linux
DISTRIB_REVISION='r47799'
root@113-76:~# olsrd -v
 *** olsr.org - 0.9.0.3-git_1bf0a71-hash_97c1ddaee6c52592843140bbbf5a8613 ***
 Build date: 2015-12-01 13:13:29 on freifunkvm

glycoknob, 2015-12-09:
I can confirm this bug, happens for me only on trunk not CC - iwdata is empty - looks like it's permission related because ubus acl's are missing or something like that

jow-, 2015-12-09:
Its not due to ubus acls but due to the fact that the olsr package used non-public internals of luci.model.networ

jow-, 2015-12-09:
Should be fixed with 63155e9. If not please reopen.

closed 2015-12-09 by jow-