openwrt/luci #570: luci-mod-freifunk: Error 500 in trunk

glycoknob, 2015-12-09:
```
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute function dispatcher target for entry '/'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: The called action terminated with an exception:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute function dispatcher target for entry '/freifunk'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: The called action terminated with an exception:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute template dispatcher target for entry '/freifunk/index'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: The called action terminated with an exception:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'freifunk/index'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/bootstrap/header'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: A runtime error occured: [string "/usr/lib/lua/luci/view/themes/bootstrap/hea..."]:78: attempt to index local 'boardinfo' (a nil value)
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: stack traceback:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]:       [C]: in function 'assert'
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]:       /usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

Problem goes away if you give ubus more privs - this is the default and doesn't work: 

```
root@netbox:/usr/share/rpcd/acl.d# cat unauthenticated.json 
{
        "unauthenticated": {
                "description": "Access controls for unauthenticated requests",
                "read": {
                        "ubus": {
                                "session": [
                                        "access",
                                        "login"
                                ]
                        }
                }
        }
}
```

looks like ubus calls are executed in a different way and need explizit access rights. 

to reproduce: http://<routerip>/cgi-bin/luci//freifunk/index

if I give more permissions in unauthenticated.json it works for me.

jow-, 2015-12-09:
Phase one of the fix has been committed with https://dev.openwrt.org/changeset/47816 - this corrects the permission handling of ubus which led to this regression in the first place.

Phase two will switch the current root ubus connection to the less privileged user context and install a proper ACL definition but I defer this change for now since it would cause the unpatched OpenWrt ubusd (prior to r47816) to segfault.

thuehn, 2016-01-11:
We do face the same problem. Is there a workaround until Jow enters his Phase II ? 

Greetings Thomas

jow-, 2016-01-11:
Please see if 81e80c4b876e8e68bb8b022c39d0941e2c1ccb56 solves your issue.

thuehn, 2016-01-11:
Perfecto ... thx Jow ... all working now in Luci

jow-, 2016-01-11:
Thanks for testing!

closed 2016-01-11 by jow-