openwrt/luci #571: luci-mod-freifunk: hostname on front page missing

FreifunkUFO, 2015-12-09:
on some public freifunk status luci pages the HTML-Title is incorrect: One "?" is printed instead of the Hostname

http://10.61.234.76/cgi-bin/luci/   -> HTML-Title:  ? - Overview - LuCI
http://10.61.234.76/cgi-bin/luci//freifunk/   -> HTML-Title:  ? - Overview - LuCI
http://10.61.234.76/cgi-bin/luci//freifunk/status -> HTML-Title: 234-76 - Status - LuCI

its also a ? on "Overview", "Contact", "Map" & "Services"

happens only on openwrt-trunk since some months..

jow-, 2015-12-09:
Same cause as #570

jow-, 2015-12-09:
Please follow up on #570.

closed 2015-12-09 by jow-

FreifunkUFO, 2016-11-04:
problem is still existing, we also track it here: https://plan.leipzig.freifunk.net/issues/361

FreifunkUFO, 2017-01-11:
still existing.

but, the hostname (missed in title-header) is mentioned inside the standard-freifunk-text at this index-page correctly