openwrt/luci #576: NullRef if luci.config.sauth.sessiontime not set

SamJongenelen, 2015-12-15:
in [luci-mod-rpc issue](https://github.com/openwrt/luci/blob/69fd4f14eef783076b69315c71fe57baad1f8925/modules/luci-mod-rpc/luasrc/controller/rpc.lua#L55) `luci.config.sauth.sessiontime`
 line # 55 is not set

This causes [this issue](https://github.com/openwrt/luci/issues/427)  (which had been closed, but still wanted to link it for the cause) and my issue [over here](https://github.com/balloob/home-assistant/issues/753)

Can someone please make a nullcheck / array check? I can't seem to get luci running in my visual studio

jow-, 2015-12-15:
I cannot reproduce the problem, that variable is provided indirectly by the "luci.config" module which maps /etc/config/luci as proxy table into the block scope.

What LuCI version did you use exactly when you encountered the problem?

jow-, 2015-12-15:
Nevermind, I see that revision 3836b45 was mentioned in the other report.
The issue described here has already been fixed with fff7c6c5188ff7a82b5d9f46a6f2c143288b5ed1.

jow-, 2015-12-15:
I also cherry-picked fff7c6c5188ff7a82b5d9f46a6f2c143288b5ed1 into the `for-15.05` branch now.

closed 2015-12-15 by jow-

SamJongenelen, 2015-12-16:
Excellent I'll pull the update tonight thanks

cctv838126, 2018-03-17:
my router is Lenovo newifi mini，i use “opkg install rpcd-mod-file -nodeps“” install the rpcd-mod-file，And create a read-only user to be used by setting up the ACL file /usr/share/rpcd/acl.d/user.json，restart the services，but the “ubus list | grep file” give no return。the homeasisstant gives no error，nether i can not find the device in the “history”。

marine1988, 2018-11-20:
I cant conect my openwrt tp with homeassistant

flipreverse, 2019-01-26:
Believe it or not: This issue is still present in OpenWRT r9136-f4d6e8f98f.
I do not see why fff7c6c is a fix to this. Why does a "require .." make things better?

cagnulein, 2020-09-07:
> Believe it or not: This issue is still present in OpenWRT r9136-f4d6e8f98f.
> I do not see why [fff7c6c](https://github.com/openwrt/luci/commit/fff7c6c5188ff7a82b5d9f46a6f2c143288b5ed1) is a fix to this. Why does a "require .." make things better?

did you fix it in some way?

Linx-ESP, 2026-10-11:
Still present it seems