openwrt/luci #610: luci-app-openvpn stores client keys in path that is not backed up

xypron, 2015-12-24:
I created a new OpenVPN client configuration "jan_client".
I then uploaded keys and certificates. Now in /etc/config/openvpn I find the following lines:

```
        option ca '/lib/uci/upload/cbid.openvpn.jan_client.ca'
        option key '/lib/uci/upload/cbid.openvpn.jan_client.key'
        option cert '/lib/uci/upload/cbid.openvpn.jan_client.
```

When I create a configuration backup directory /lib/uci/upload is not included.
Shouldn't the keys and certificates be stored somewhere in the /etc path, eg. /etc/openvpn/keys?

xypron, 2015-12-24:
https://dev.openwrt.org/ticket/21437

hnyman, 2016-01-02:
Most likely fixed in trunk by https://github.com/openwrt/luci/pull/562

closed 2016-01-02 by xypron