openwrt/luci #611: Luci ddns problem

mauriceklein, 2015-12-27:
Hello,
if i try to safe my new ddns configuration by clicking the save button,i get this error:

/usr/lib/lua/luci/dispatcher.lua:452: Failed to execute cbi dispatcher target for entry '/admin/services/ddns/detail/myddns_ipv6'.
The called action terminated with an exception:
/usr/lib/lua/luci/tools/ddns.lua:170: unhandled validate error
stack traceback:
[C]: in function 'assert'
/usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
/usr/lib/lua/luci/dispatcher.lua:141: in function

I reeinstalled the Luci ddns package and also the ddns scripts package and i flashed the whole router with openwrt and i didn't keep the settings but nothing helped.

I've installed luci-app-ddns 2.2.4-1 and ddns-scripts 2.5.0-2.

chris5560, 2015-12-27:
Hi, 
could you please edit /usr/lib/lua/luci/tools/ddns.lua at line 170 for testing
OLD: 	assert( not (errtxt and (#errtxt > 0)), "unhandled validate error" )
NEW:	assert( not (errtxt and (#errtxt > 0)), "unhandled validate error: " .. errtxt )

and send me the error-message.
Do you have any idea what value you set, might produce the problem ?
Which settings of the default configuration did you change ?

To disable later on the error edit line 170 again
OLD:	assert( not (errtxt and (#errtxt > 0)), "unhandled validate error" )
NEW:	-- assert( not (errtxt and (#errtxt > 0)), "unhandled validate error" )

Now you no longer get the error message inside LuCI, but the problem might now moved to ddns-scripts.
Either one of your settings might be wrong or the validate function for one value inside luci-app-ddns produce error.

Thanks
Christian

mauriceklein, 2015-12-27:
Hello,
I changed the line but i got the same error text:

/usr/lib/lua/luci/dispatcher.lua:452: Failed to execute cbi dispatcher target for entry '/admin/services/ddns/detail/noip'.
The called action terminated with an exception:
/usr/lib/lua/luci/tools/ddns.lua:170: unhandled validate error
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

And also when i edeted the line with -- assert( not (errtxt and (#errtxt > 0)), "unhandled validate error" ) i get the same page.
My idea is that i set the Lookup Hostname false, because i set it only to my ddns hostname and domain xxxxxx.ddns.net because i thinked that this is for a lookup if the ip adress is right but i am not sure.

chris5560, 2015-12-28:
Ups,
I think I forgot to tell you that after every edit you need to clear browsers cache and remove /tmp/luci-indexcache file and possibly existing /tmp/luci* directories on the router. Otherwise the changes to the ddns.lua are not recognised.
Sorry. Could you please recheck.
Thanks
Christian

mauriceklein, 2015-12-29:
Ok it worked.
The Errortext is:
/usr/lib/lua/luci/dispatcher.lua:452: Failed to execute cbi dispatcher target for entry '/admin/services/ddns/detail/noip'.
The called action terminated with an exception:
/usr/lib/lua/luci/tools/ddns.lua:170: attempt to concatenate local 'errtxt' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

PR #616 (2015-12-31): luci-app-ddns: removing unneeded assert
- remove unneeded assert() in tools/ddns.lua fix #611
- set PKG_VERSION to the correct value from controller/ddns.lua
closed 2015-12-31 by chris5560