openwrt/luci #618: cbi.lua: Problem handling optional options inside Map.parse()

chris5560, 2016-01-03:
Inside luci-app-privoxy and luci-app-openvpn and possibly other apps, values with ".optional" set true are not parsed correctly.
Inside OpenWrt 15.05 everything works fine.
[image]
I did not find the pull where Map.parse() was rewritten inside cbi.lua, but in current trunk
when you start the same page the optional values are missing.
[image]
If you press the [SAVE] button once (without any changes) the optional values are shown.
[image]

If you set sections property dynamic to true it's nearby the same.
After first call of the page you get an empty field to add an option but no selection list.
After pressing [SAVE] everything works as expected.

I crosscheck and only replace Map.parse() function inside cbi.lua of current trunk and everything works fine. (Screenshots are take from luci-app-privoxy)

chris5560, 2016-01-10:
@jow- could you please have a look at ?
To see problem on luci-app-openvpn do the following:
Create a new configuration (client ethernet bridge)
On the opened page select remove on option "verb" and "comp_lzo"
Press [Save & Apply] and go back to overview
go back and edit your configuration and all optionals are missing
Press [Save] and the selection list to add optionals is back

jow-, 2016-01-10:
Will take a look soon but I'll rather remove optionals functionality entirely.

chris5560, 2016-01-10:
Why ?

chris5560, 2016-01-13:
@jow
I found additional problems with new parse function:
If using custom validate function because .datatype for an option doesn't fit:
Old: If you enter an invalid value and press [Save]/[Save&Apply] no data is written to uci.
New: data of other fields changed is written to uci (pending updates) and 
really a problem, if [Save & Apply] the data except the invalid is written to uci and service is restarted.

If the list of options shown on one screen is long enough, nobody will see the error message at the bottom. because screen roll up after pressing one of this action button.

jow-, 2016-01-15:
@chris5560: I do not like optionals because they're complicating the code and are counter-intuitive to use. Take an optional boolean for example: you first have to add it (which triggers an http request/response cycle already) then you have to set it (after you found out where exactly it popped up in the form), then you have to save again (next http request/response).

Optionals are especially confusing if they refer to default enabled boolean options. Take for example "option enabled 0" marked as optional - you first have to "add" the enabled option (despite the fact that it is enabled already) only to "remove" (uncheck) it once it appeared.

It is also confusing in the case of text input fields that suddenly disappear if made empty.
For new applications I rather suggest to move those optionals into an "advanced" tab or something.

chris5560, 2016-01-15:
@jow
OK good arguments. When I look at luci-app-radicale next time I will try to remove optionals.
I also need to change radicale package.
One reason why I use optionals was, because AbstractValue.size property is not handled by any StyleSheet.
It's defined in the corresponding views and set to nil (why) inside AbstractValue but changing it changes nothing.

By the way. Do you see a possibility to add tabs to maps not only to sections.
Looking on luci-app-radicale there are multiple sections that should be split into separate tabs.
One of these sections is currently handling the "optional" options.
Do you have a code sniplet how to realise one map and each section on a separate tab ?

jow-, 2016-01-15:
@chris5560 I'll see if I can easily add map-level tab support.

jow-, 2016-01-15:
The original issues should be solved now btw.

closed 2016-01-15 by jow-

jow-, 2016-01-15:
@chris5560 I added very basic support for arranging the section objects as tabs in maps. Simply set `m.tabbed = true` to enable it. Haven't done any extensive testing or css adjustments for this yet but it appears to work okay mostly.

chris5560, 2016-01-15:
@jow
map.parse fix works
m.tabbed I will check
Thanks