openwrt/luci #620: ddns error

ezar, 2016-01-09:
/usr/lib/lua/luci/dispatcher.lua:452: Failed to execute cbi dispatcher target for entry '/admin/services/ddns/detail/myddns_ipv4'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:458: bad argument #4 to 'set' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

chris5560, 2016-01-09:
@ezar
I cannot see which option inside luci-app-dddns produce the error.
Could you please change one option after the other with pressing the [SAVE] button after each changed option.
By doing so you should find the option producing the error.
Please come back with the option and the value you try to set, so I can look into the code.
Many thanks for your support.

ezar, 2016-01-09:
No option changed. I enter in cgi-bin/luci/admin/services/ddns/detail/myddns_ipv4 and press SAVE.

Using:
OpenWrt Designated Driver r48147 / LuCI Branch (git-16.004.80366-03a8393)

Thank you

hnyman, 2016-01-13:
I got the same issue when I tried editing the options. Up-to-date trunk.
Starting from defaults, no DDNS config saved earlier.

hnyman, 2016-01-13:
This might be related to optional option values (and https://github.com/openwrt/luci/issues/618 )
Looks like the function tries to parse a non-existing argument.

    /usr/lib/lua/luci/cbi.lua:458: bad argument #4 to 'set' (string expected, got nil)

Line 458 is inside Map.set:
https://github.com/openwrt/luci/blob/31816e92f109e0c534f176ca1e79b032917c876c/modules/luci-base/luasrc/cbi.lua#L458

EDIT:
I found at least one culprit: "Use HTTP Secure" option.
* That option enabled --> save works
* That option disabled --> save fails, the error above

chris5560, 2016-01-13:
I'm working on. please give some days because fixing one brings up new problems.

jow-, 2016-01-17:
Is this still an issue after the optionals fixing?

chris5560, 2016-01-18:
@jow
I don't know exactly.
But it's a new version including some fixes prepared, waiting for new ddns-scripts 2.6.0-1 version to be merged.
Thanks
Christian

closed 2016-02-07 by chris5560