openwrt/luci #623: Discussion on Luci multi-user features

hnyman, 2016-01-14:
@Hostle has created a functional system for enabling handling multiple users in Luci.
* User creation, removal and provisioning work.
* Two user classes: admin and user. (root has always full access.)
* Possibility to adjust access on menu-item level 
* Possibility to deny ssh access

@Hostle has actually created two alternative methods for a year-old Luci (from March 2015, CC15.05 development cycle) into his Github repo. I have imported the changes into up-to-date DD trunk and have also squashed and polished the commits for discussion purposes to have minimal diffs.

Hostle thinks that the newer method is better, but I will present both here. That serves the discussion by highlighting the various possibilities.

Original discussion at forum: https://forum.openwrt.org/viewtopic.php?id=54593

"Old method":
======
"Old" original repo with ~60 commits:  https://github.com/Hostle/openwrt-luci-multi-user/commits/master
"Old" from my repo, with 3 commits, "multiuser" branch:
https://github.com/hnyman/luci/commits/multiuser

The old method requires unique page/leaf names, so one of the commits is just for renames of existing pages in status and network menus.

"New method":
======
"New" original, 14 commits from Hostle, CC files as originals instead of DD for the modified files, so it does not actually work with otherwise DD trunk Luci): https://github.com/Hostle/luci

"New" from my repo, "multi2" branch, 12 commits preserving the explanations from Hostle, pure DD stuff, works:
https://github.com/hnyman/luci/commits/multi2

I updated the CC files int DD versions (CRSF tokens etc.) and edited the files so that the diff shown in Github is minimal, in order to highlight the actual changes (instead of indentation changes etc.). My version has 12 commits and a file is touched only by one commit, so it is very clean.

"new" does not require renaming existing pages.

Identified problems/challenges:
=====

* Both methods have the drawback that the permission settings only work for previously known pages or menu tabs. E.g. new page system / Custom commands from luci-app-command is not handled. It would be better, if the system somehow sniffed the pages and then decided the permissions based on the main menu tab.
* "new" method required editing existing pages to have permission-checking wrappers around each menu item. E.g. in status, system and network. E.g. https://github.com/hnyman/luci/commit/e0c74029bedca82453b86a59a11c156aa4fcf6dd
* "old" method requires each page to have unique name. Currently e.g. both status and network have a page with name "routes". Having unique page names might be useful later for other purposes, so I think that this change might be considered in any case.
* I compiled "new" for my own ar71xx build and it worked to large extent, but not quite fully with Firefox. Might be cache problems or something. Hostle himself compiled "new" from my repo to his trunk and says that it worked perfectly.
https://github.com/Hostle/luci/commit/c1ba7d780fbafba1d882e3d05ff96baeaee131f3#commitcomment-15451559

Trying the code:
=========
If you want to try the multi-user code, you can easily add my Luci repo as a remote to your own git and then pull from "multi2" branch. I used these commands to import this to my own Openwrt build. (I also created a new branch "multiuser" at my local feed repo so that I easily push the changes aside by "git checkout master"):
```
 cd feeds/luci
 git checkout -b multiuser
 git remote add hnyman https://github.com/hnyman/luci.git
 git pull hnyman multi2
 git log --oneline
```
The multi2 branch contains up-to-date Luci of 11 Jan 2016.

Screenshot from trunk Designated Driver r48235:
[image]

@jow- 
Hopefully you have time to check a bit into this idea and give @Hostle suggestions for further development.

hnyman, 2016-01-14:
I think that I spotted a major flaw in the "new" method:
the definition of allowed pages are cumulative for users. One page allowed for at least one user in group "user" is allowed for all users in group "user".

I defined two limited users "koe" and "test" and allowed them different pages in the status tab. Still both saw any page that at least one of them saw:
```
root@OpenWrt:~# cat /etc/config/users

config user
        option name 'koe'
        option user_group 'user'
        option shell '1'
        option Status_menus 'Status_menus'
        option Network_menus 'disabled'
        option System_menus 'System_menus'
        option system_subs 'Software'
        option status_subs 'Firewall Routes'

config user
        option name 'test'
        option user_group 'user'
        option shell '0'
        option System_menus 'disabled'
        option Network_menus 'Network_menus'
        option network_subs 'Switch Firewall'
        option Status_menus 'Status_menus'
        option status_subs 'System_log Kernel_log'
```
[image]

The rights updated ok on the fly when I adjusted rights in one firefox browser with admin rights and used one IE window as plain limited user. So, the provisioning itself worked ok, but the user just got too much rights from the other basic users.

I guess that the allowed pages provisioning should be by the user class. E.g. users or admin. And each user should belong into into exactly one group. Then this might work.

Hostle, 2016-01-14:
i replicated your error and tracked down the source, thankfully its just an
error in the code...not in the logic :) I will make the necessary changes
tonight

Hostle, 2016-01-15:
I have fixed the global permissions issue with the following commits ...

https://github.com/Hostle/luci/commit/987b4b5809fa5215f7e8221384dedeae2765ef7d
https://github.com/Hostle/luci/commit/2574d144c925c31301d64ea6a0c6a0887e2aadbb
https://github.com/Hostle/luci/commit/b6c82e56e9d7c8e089e50be9e8125242fb00be20
https://github.com/Hostle/luci/commit/e9e4115533190e46c83aa6f2c0b3a51fb3eda5d4
https://github.com/Hostle/luci/commit/8cf860c8b0ac15650f686b77b0d686504f8d476f

Please test it and let me know if it has fixed the issue for you, it works
on my end :)

Hostle, 2016-01-15:
I cleaned up the user module, I removed old redundant functions and cleaned
up code for easier reading and editing

https://github.com/Hostle/luci/commit/d5468a451b31781247240bcffa23735b971c324e

hnyman, 2016-01-15:
You are right, the rights work ok after those changes. Each user has separate rights. I will incorporate those into my source tree.

However, I noticed a new bug:
if you enable a tab but do not enable any pages, you will get this error:
```
usr/lib/lua/luci/users.lua:292: attempt to concatenate a nil value
stack traceback:
	/usr/lib/lua/luci/users.lua:292: in function 'get_menus'
	/usr/lib/lua/luci/users.lua:303: in function 'hide_menus'
	/usr/lib/lua/luci/controller/admin/network.lua:23: in function 'user'
	/usr/lib/lua/luci/controller/admin/network.lua:33: in function 'v'
	/usr/lib/lua/luci/dispatcher.lua:552: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:216: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```
That was after user "test3" had access to Network menu but he had no pages enabled.

Enabling at least one page returns Luci to the normal functionality.
```
config user
        option name 'test3'
        option user_group 'user'
        option shell '1'
        option System_menus 'disabled'
        option Network_menus 'Network_menus'
        option Status_menus 'Status_menus'
        option Status_subs 'Firewall'
```
Add ``` option Network_subs 'Wifi'``` and it works again.

hnyman, 2016-01-15:
I also noticed that when users with limited right visit luci pages, system log gets messages for each page:
```
Fri Jan 15 10:50:15 2016 daemon.err uhttpd[2276]: uci: Entry not found
Fri Jan 15 10:50:16 2016 daemon.err uhttpd[2276]: uci: Entry not found
Fri Jan 15 10:50:21 2016 daemon.err uhttpd[2276]: uci: Entry not found
Fri Jan 15 10:50:26 2016 daemon.err uhttpd[2276]: uci: Entry not found
Fri Jan 15 10:50:31 2016 daemon.err uhttpd[2276]: uci: Entry not found
Fri Jan 15 10:50:33 2016 daemon.err uhttpd[2276]: uci: Entry not found
```
There seems to be at least one line for each visited page.
Additionally, each dynamically updating (every 5 secs) XHR component generates a new error line to the log.

jow-, 2016-01-15:
The uci errors in the log are due to the fact that host directly parses "uci show users" output using "io.popen()" - this is extremely inefficient - please use the uci bindings directly.

Hostle, 2016-01-15:
I will rewrite things to use the uci binding, thanks Jow

Hostle, 2016-02-12:
I think we need to revisit the conversation, I have revamped the code to use uci bindings and redid the entire interface in a much cleaner way to work with the new backend. I have added all of the suggestions made by Jow. I agree  about using nixio to add and remove shadow and passwd entries and I would add the perhaps the set user shell should be done in nixio as well, something similar to chsh. I have applied all of the latest updates to my hub here https://github.com/Hostle/luci/tree/master/applications/luci-app-multi-user

hnyman, 2016-02-23:
Somebody asked a question about the multi-user code in the forum a few minutes ago, just before the forum crashed once again. I have imported the changes from hostle until 11 days ago to the new multi3 branch, and I have just rebased that branch to the top of the current Luci trunk.
https://github.com/hnyman/luci/commits/multi3

That should match hostle's multi-user code with the current trunk.

Hostle, 2016-02-23:
your multi-user commits are way behind mine.... check out my latest commits

hnyman, 2016-02-23:
Looks like that. I was just browsing that and noticed that you have added new commits ~2 weeks ago, apparently after I first created the multi3 branch.
I have likely your January commits, but not those from February.
(and it is a bit hard to identify them from the ftp stuff etc. as they all are in the same branch)

Hostle, 2016-02-23:
yeah i need to clean up, i will tidy things up this evening, I am still
relativity new to git so I am still learning. All commits from February 7
to February 12th

hnyman, 2016-02-23:
I imported your changes to Luci trunk tip, and compiled.
At first the creation of new users did not work at all.
But then I spotted the filename error at https://github.com/Hostle/luci/blob/master/applications/luci-app-multi-user/root/etc/uci-defaults/09_users (due to commit https://github.com/Hostle/luci/commit/6d280f2dacdd148cde300ac587ab84a4ae283ca8 and the empty fix??? of https://github.com/Hostle/luci/commit/44f6f917567bcf835ad19c2a9a7689384485b1e2 . Strange. Some push error to Github?) 

I pushed a fix for that in my repo and re-compiled.
Not editing users works.

But when a new user has been created, you can't see the possible menu subitems on the first visit to the user. Instead you have to save him first and then at the next visit the items may get visible. 

That is possibly due to the recent changes in Luci trunk. (your repo is 88 commits behind the Luci trunk.)

Hostle, 2016-02-23:
> (due to commit Hostle@6d280f2
> and the empty fix??? of Hostle@44f6f91

Excellent, I will look into the issue this evening and try to correct it.

Thanks Hannu

Hostle, 2016-02-23:
It appears to be a problem with the multi value dependency handling, the
Network>>Interfaces>>Add Interface is broken aswell, the check box
Create a bridge over multiple interfaces

Should enable the "Cover the following interfaces" list, this check box
currently does nothing and the multiValue list is always visible. I think
this warrants a ticket

hnyman, 2016-02-24:
> It appears to be a problem with the multi value dependency handling, the Network>>Interfaces>>Add Interface is broken as well, the check box Create a bridge over multiple interfaces 

Yeah, I can see the problem at https://192.168.1.1/cgi-bin/luci/admin/network/iface_add 

This sounds like something similar as the issue in https://github.com/openwrt/luci/issues/648#issuecomment-187338044 , which was fixed to some extent by https://github.com/openwrt/luci/commit/385a49277ac3339e3ea9a41abde2db0eba5427f9 . Apparently there are similar issue still left. That will require help from @jow-

Hostle, 2016-02-24:
> This sounds like something similar as the issue in #648 (comment)
> which was fixed to some extent by 385a492

Yes, it's very similar to the #648 issue, going through the latest commits
I see there has been many changes to the cbi functionality and many
problems as a result. I will try to dig into it a little further and see if
I can find out which commit broke it

Hostle, 2016-02-24:
I have found the solution for the problem with the multivalues not
appearing, this fixes the  users page but this does not fix the issue with
the https://192.168.1.1/cgi-bin/luci/admin/network/iface_add
page, tho I suspect this this was something different since the issues were
not the same

here is the corrected mvalue.htm that fixes our issue...

[code]
<%
    local i, key
    local v = self:valuelist(section) or {}
-%>

<%+cbi/valueheader%>
<% if self.widget == "select" then %>
    <select class="cbi-input-select" multiple="multiple" data-update="click
change"<%=
        attr("id", cbid) ..
        attr("name", cbid) ..
        ifattr(self.size, "size")
    %>>
        <% for i, key in pairs(self.keylist) do -%>
            <option<%=
                attr("id", cbid.."-"..key) ..
                attr("value", key) ..
                attr("data-index", i) ..
                attr("data-depends", self:deplist2json(section,
self.deplist[i])) ..
                ifattr(luci.util.contains(v, key), "selected", "selected")
            %>><%=pcdata(self.vallist[i])%></option>
        <%- end %>
    </select>
<% elseif self.widget == "checkbox" then %>
    <div>
        <% for i, key in pairs(self.keylist) do %>
            <label<%=
                attr("for", cbid.."-"..key) ..
                attr("data-index", i) ..
+              attr("data-depends", self:deplist2json(section,
self.deplist[i]))..
+              ifattr(luci.util.contains(v, key), "checked", "checked")

            %>>
                <input class="cbi-input-checkbox" type="checkbox"
data-update="click change"<%=
                    attr("name", cbid) ..
                    attr("value", key) ..
                    ifattr(luci.util.contains(v, key), "checked", "checked")
                %> />
                <%=pcdata(self.vallist[i])%>
            </label>
            <% if i == self.size then write('<br />') end %>
        <% end %>
    </div>
<% end %>
<%+cbi/valuefooter%>
[code]

I have put a + in front of the 2 lines I edited. If you could pass the
message along to Jow or point me in the right direction I can post the code
and explain the issue

Hostle, 2016-02-24:
I opened up an issue here -->> https://github.com/openwrt/luci/issues/652,
helpfully that helps Jow to find a solution

jack338c, 2016-05-12:
@hnyman 

Hello,I'm a new guy,please help me

after done the following code,

 cd feeds/luci
 git checkout -b multiuser
 git remote add hnyman https://github.com/hnyman/luci.git
 git pull hnyman multi3
 git log --oneline

then I do :./scripts/feeds update -a ,./scripts/feeds install -a ,make menuconfig ,make -j 3

but the image did not have the mutluser function.How can I make this works.?

thanks

Jack

Hostle, 2016-05-12:
you need to select the multi-user package during the menuconfig ...

LuCI >> Apps >> Multi-user

jack338c, 2016-05-12:
@Hostle 
Hello,Hostle
I did it according your way,but it does't  work yet.
I only find LuCI >> Apps>luci-app-mutlwan something else 
please help me 
thanks

hnyman, 2016-05-19:
@jack338c 
You need to use the Openwrt trunk (Designated Driver), not the previous 15.05 branch.
I followed your example and it worked just great...
```
  cd feeds/luci
  git checkout -b multiuser
  git remote add hnyman https://github.com/hnyman/luci.git
  git pull hnyman multi3
  git log --oneline
  cd ../..
  ./scripts/feeds update
  ./scripts/feeds install -a

And note that the feeds install phase notifies about the new luci-app-multi-user, after which it will be available in menuconfig:
perus@ub1604:/Openwrt/trunk$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing package 'luci-app-multi-user' from luci
Installing all packages from feed routing.
...
```

I have also updated the "multi3" branch to the current head of Luci git, (May 2016).

tcpipchip, 2016-05-19:
Sir, can you give us support to port it to CHAOS 15.05.1 ?
Let me know the cost.

Hostle, 2016-05-19:
I will take a look on the weekend, the original support was based on CC 15.05, it is available in my repo but I may need to update the src with the latest changes and test with 15.05.1 once I have tested it I will post a link to the repo .. free of charge of course, that's the whole idea of opensrc projects :) contribute where you can! 

Cheers

tcpipchip, 2016-05-19:
Nice!!!!!, My contribute is to test it on LINKIT SMART 7688 for you!!!!
btw, 
cd feeds/luci
  git checkout -b multiuser
  git remote add hnyman https://github.com/hnyman/luci.git
  git pull hnyman multi3
  git log --oneline
  cd ../..
  ./scripts/feeds update
  ./scripts/feeds install -a

is the instruction to install ?

jack338c, 2016-05-19:
@hnyman 
thank you very much for your help.it works for me now
give my best with you !

is that possible to hide the service menu in openwrt? if so,that's will be more attractive ！

tcpipchip, 2016-05-20:
Hostle,

 **"...the original support was based on CC 15.05, it is available in my repo but I may need to update the src with the latest changes and test with 15.05.1"**

Did you test it o CC 15.05 and works perfectly ? I had tested on linkit but i have problems when i access some menu that not exists.

Would you like to make some tests in my board ?

hnyman, 2016-05-21:
> Is that the sequence install that you use in the CC 15.05 for install UI-MULTI-USER ?
No.
The code in my github repo (hnyman) is for trunk only.

There is no cc15.05 version of the current code. (hostle originally made the code for cc15.05, but then modified it to match the changes in trunk as that is where the development should be done.)

tcpipchip, 2016-05-22:
Ok! Thank you so much!

Enviada do meu iPhone

tcpipchip, 2016-07-19:
Hello, is there some update that allow multiusers have access to "Mount Point" and "MJPG-streamer" menus to enable them or not ?
Thanks so much!

Hostle, 2016-07-19:
I am just finishing up the code that will allow  all luci addon meus to be
enabled/disabled by the root user ... I have it functioning now, I am just
in the process of debugging and uploading the new code to my git, then if
you ask Hannu really nice he may just upload it to his git too. I hope to
have it finished before the weekend.

Hostle

tcpipchip, 2016-07-19:
Hi, did you read my mind ??? [😊]

Please, let me know when ready! Ok ?

Miguel

Hostle, 2016-07-20:
You can test it out if you replace you luci feed with mine repo in your
feeds.conf.default ...

src-git luci https://github.com/Fire-Wrt-2016/luci;luci-multi-user-15.05.1

then do

 ./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig

then select the multi-user package in the LuCI-->Applications -->>
luci-app-multi-user

It should work fine, if you have any problems let me know. Note: this is
for chaos_calmer 15.05.1 branch, I am not quite done the trunk version yet.

Hostle

tcpipchip, 2016-07-20:
Nice, i will test...

Btw, i am using today your multi-user in a project...all working fine...except this update...

If i created a IPK, may i install OVER that version ?

Miguel

Hostle, 2016-07-20:
No you must do a complete update, most of the new code is in the main dispatcher.lua file

tcpipchip, 2016-07-20:
I will test now and see if all works!

tcpipchip, 2016-07-20:
Hi,  i found some bugs.

Btw, did you test on wich design, openwrt, freifunk, bootstrap ?

Let me know!

Then i can report the bugs

Hostle, 2016-07-20:
I only use bootstrap.. however the code should not be affected by th theme

tcpipchip, 2016-07-20:
lets go!

1) If you Enable Network Menus and only select Interfaces, when you open browser, you will see Interfaces and Firewall too!

2) if you Enable System Menus and only select Mount_Points when you open browser, you can´t see the System Menus, but, if you choose Mount_Points and System, you will see the System Menus and Mount Points  and System

See

https://dl.dropboxusercontent.com/u/83554849/bug1.png

<https://dl.dropboxusercontent.com/u/83554849/bug1.png>I can only see System Menu if i choose System item and Mount Points

[https://dl.dropboxusercontent.com/u/83554849/bug1.png]

Do you understand ?

Miguel

tcpipchip, 2016-07-20:
https://dl.dropboxusercontent.com/u/83554849/bug2.png

For example, to get access to Mount Point, i had to enable System too [☹]

[https://dl.dropboxusercontent.com/u/83554849/bug2.png]

Hostle, 2016-07-20:
Ok that makes sense, I am guessing it it because the System main tab and the System--»System are being processed as one. Should be an easy fix.

tcpipchip, 2016-07-20:
Let me know when you fix it! Ok ?

Thanks so much!

Enviada do meu iPhone

Fire-Wrt-2016, 2016-07-21:
This issue should be fixed with latest commits, rebuild and try again ...

tcpipchip, 2016-07-21:
Hello!

I will test now!

Thanks so much!

Miguel

tcpipchip, 2016-07-21:
Now looks that is working!!!!!!

Congratulations!

A friend will test too soon!

I will report!

RJRvdMeer, 2016-09-28:
Hello,

What "src-git luci " should I use, if I want the latest multi-user and material thema?
Or can you help me to get this running? 

I tried the following in feeds.conf.default
# src-git luci https://github.com/openwrt/luci.git
# src-git luci https://github.com/openwrt/luci.git;for-15.05
# src-git luci https://github.com/Hostle/openwrt-luci-multi-user.git
src-git luci https://github.com/Fire-Wrt-2016/luci

Robert

tcpipchip, 2016-09-28:
https://github.com/Hostle/openwrt-luci-multi-user.git

RJRvdMeer, 2016-09-28:
Thx tcpipchip. Tried it, and multi user works. Only no material theme.

tcpipchip, 2016-09-28:
change the theme and will work!

RJRvdMeer, 2016-09-28:
sry, I must do something wrong. when I update and install luci I don't have material theme anymore as een option in LuCI->Themes.

hnyman, 2016-09-28:
You need to install the material theme separately. That has nothing to do with multiuser.

RJRvdMeer, 2016-09-28:
ah ok. any tips on that? just since material theme is in the openwrt/luci git and I'm not sure how to fetch this in a correct way. (in stead of copying it from this git)

hnyman, 2016-09-28:
> > You need to install the material theme separately. 

> ah ok. any tips on that? j

If trunk are running trunk just do
```
opkg update
opkg install luci-theme-material
```
Or do the same actions in Luci (System -> Software).

tcpipchip, 2016-09-28:
i tested on chaos and works too

RJRvdMeer, 2016-09-29:
Update by software in router menu is working. System->software. Thx :)

Is there a way to get the option in menuconfig? Love to learn this.

RJRvdMeer, 2016-10-04:
@hnyman 

Is there a way to get luci-theme-material in menuconfig? 'Cause it isn't there after update and install, using src-git luci https://github.com/Hostle/openwrt-luci-multi-user.git in feeds.conf.default

hnyman, 2016-10-04:
> Is there a way to get luci-theme-material in menuconfig? 'Cause it isn't there after update and install, using src-git luci https://github.com/Hostle/openwrt-luci-multi-user.git in feeds.conf.default 

Well, whole that repo is very old LuCI code that has not been updated since Oct 2015. That repo does not even contain the current Luci (for DD trunk), so there is no Material theme. 

Hostle worked actively with multi-user in Jan-Feb 2016 using https://github.com/Hostle/luci , which commits I have imported into my own repo (see the first post), fixed the uci-defaults error (still there in Hostle's repo) and rebased every now and then to the top of the current Luci repo. I have not tested that code in 6 months, but that might still be your best bet if you are using Openwrt DD trunk (or LEDE).

```src-git luci https://github.com/hnyman/luci.git;multi3 ```

That is current LuCI DD trunk / LEDE plus the multi-user components for DD trunk from hostle as of Feb 2016.

The developments during this summer for 15.05.1 (discussed between Hostle and tcpipchip) are probably in https://github.com/Fire-Wrt-2016/luci/tree/luci-multi-user-15.05.1 , but I haven't checked the exact nature of the changes.

Luci for DD trunk & LEDE has some differences compared to Luci for CC15.05, so the CC code might not run on DD and vice versa.

Hostle, 2016-10-04:
I will look at updating the current trunk with the new commits made to the CC code for the services menus.                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone on the Rogers network.                                                                                                                                                                                                                From: Hannu NymanSent: Tuesday, October 4, 2016 10:22 AMTo: openwrt/luciReply To: openwrt/luciCc: Hostle; MentionSubject: Re: [openwrt/luci] Discussion on Luci multi-user features (#623)
Is there a way to get luci-theme-material in menuconfig? 'Cause it isn't there after update and install, using src-git luci https://github.com/Hostle/openwrt-luci-multi-user.git in feeds.conf.default 

Well, whole that repo is very old LuCI code that has not been updated since Oct 2015. That repo does not even contain the current Luci (for DD trunk), so there is no Material theme. 

Hostle worked actively with multi-user in Jan-Feb 2016 using https://github.com/Hostle/luci , which commits I have imported into my own repo (see the first post), fixed the uci-defaults error (still there in Hostle's repo) and rebased every now and then to the top of the current Luci repo. I have not tested that code in 6 months, but that might still be your best bet if you are using Openwrt DD trunk (or LEDE).

src-git luci https://github.com/hnyman/luci.git;multi3

That is current LuCI DD trunk / LEDE plus the multi-user components for DD trunk from hostle as of Feb 2016.

The developments during this summer for 15.05.1 (discussed between Hostle and tcpipchip) are probably in https://github.com/Fire-Wrt-2016/luci/tree/luci-multi-user-15.05.1 , but I haven't checked the exact nature of the changes.

Luci for DD trunk & LEDE has some differences compared to Luci for CC15.05, so the CC code might not run on DD and vice versa.

[180 more replies]

closed 2020-04-19 by hnyman