openwrt/luci #625: LuCI Trunk Path

gwlim, 2016-01-17:
LuCI Trunk hyperlink generates 2 '//' in full path after clicking on a few links
http://192.168.1.1/cgi-bin/luci//admin/network/wireless
should be
http://192.168.1.1/cgi-bin/luci/admin/network/wireless

hnyman, 2016-01-17:
I think that this started to happen since https://github.com/openwrt/luci/commit/79383f5a74bcff528a5270a930c36b58e310010c fixed tthe negative impact from URL token removal

jow-, 2016-01-17:
a) is there a real problem or just a cosmetical issue?
b) a description on *where* this faulty links are generated would be extremely helpful...

hnyman, 2016-01-17:
I think that it is cosmetical and is related to selecting pages rom the top menu.
I don't think that the re are any actual negative effects.

I see // after cgi-bin/luci on all Luci pages that I have entered **via the top menu**.
Like this:
https://192.168.1.1/cgi-bin/luci//admin/status/realtime
https://192.168.1.1/cgi-bin/luci//admin/status/overview
https://192.168.1.1/cgi-bin/luci//admin/network/vlan

However, if the same page **is visited via tab change, there is single /**
Example:
Select System/software from menu:  https://192.168.1.1/cgi-bin/luci//admin/system/packages
Go to "config" tab: https://192.168.1.1/cgi-bin/luci/admin/system/packages/ipkg
Go to "actions" tab: https://192.168.1.1/cgi-bin/luci/admin/system/packages

Happens both with IF11 and  Firefox 45.

jow-, 2016-01-17:
Should be fixed with b3c69b123a698e596a5df4797378d976718247f0.

closed 2016-01-17 by jow-