openwrt/luci #631: luci breaking ubusd (WDR4300)

jacobopantoja, 2016-01-22:
Hi,

After bisecting, it seems that commit 81e80c4b8 is breaking somehow ubus, and my WDR4300 does not boot correctly due to the lack of proper ubus. (I can start in failsafe and, when I try to run "ubusd", it gives a "segmentation fault").

Can I do something else to debug?
Thanks!

git bisect start
# good: [53d767de5ea8c596936fbf0bddce357efe0c4425] luci-app-statistics: Adjust ping graphs to show target hosts separately
git bisect good 53d767de5ea8c596936fbf0bddce357efe0c4425
# bad: [63d70daae1921d9f5a92cf2221ca4f1601c78ef5] luci-mod-admin-full: align assoclist in wifi overview with status index page
git bisect bad 63d70daae1921d9f5a92cf2221ca4f1601c78ef5
# good: [967bb1f36f9f30d19e13aecf3a550eea5fd73c6b] Merge pull request #464 from remakeelectric/pulls/list-operations
git bisect good 967bb1f36f9f30d19e13aecf3a550eea5fd73c6b
# good: [edf352efa27d6e936c0cc87abf0f28f1b6b28a44] i18n: sync translations
git bisect good edf352efa27d6e936c0cc87abf0f28f1b6b28a44
# bad: [1bf268de06c1697fdf08b65dcc13c06b4bed909f] luci-base: add basic support for CBI map level tabbing
git bisect bad 1bf268de06c1697fdf08b65dcc13c06b4bed909f
# bad: [81e80c4b876e8e68bb8b022c39d0941e2c1ccb56] luci-base: properly handle ubus connections for non-root (#570, #571)
git bisect bad 81e80c4b876e8e68bb8b022c39d0941e2c1ccb56
# good: [0f1e39a90ab29ba8a754c6a0636d742e7a612a02] Merge pull request #616 from chris5560/master
git bisect good 0f1e39a90ab29ba8a754c6a0636d742e7a612a02
# good: [71dc54af6affa1a0f9b26f5fbe2fe874f2791630] a cleaner solution for disabling dyngw and dyngw_plain in freifunk-policyrouting init
git bisect good 71dc54af6affa1a0f9b26f5fbe2fe874f2791630
# good: [6619e66fc9de32e7bf911b929e320ab17a414697] luci-proto-ipv6: allow hostnames for dslite peer addresses (#621)
git bisect good 6619e66fc9de32e7bf911b929e320ab17a414697
# first bad commit: [81e80c4b876e8e68bb8b022c39d0941e2c1ccb56] luci-base: properly handle ubus connections for non-root (#570, #571)

closed 2016-01-22 by jacobopantoja

myssea, 2016-05-24:
Hello jacobopantoja,
I face the same problem, may i ask how can you resolve it?
Thank you, please give me a hand!

jacobopantoja, 2016-05-25:
Hi @myssea, although it seems that you've solved it, what I whas doing wrong was that I was not updating luci from git to the latest version. I was only updating openwrt from git. When I made an update to the feeds, everything solved.

myssea, 2016-05-26:
@jacobopantoja HaHa, thanks for your reply. It's great that we both solved the problem!
Best regard for you!