openwrt/luci #632: po2lmo is missing .. probably in staging dir

thuehn, 2016-01-26:
while compiling latest dd trunk with luci, the compile process stops with the following error:

install -d -m0755 /Users/bluse/Desktop/Coding-Area/openwrt/designated_driver_trunk/build_dir/target-powerpc_8540_musl-1.1.12/luci-app-firewall/ipkg-all/luci-i18n-firewall-de/usr/lib/lua/luci/i18n
po2lmo /Users/bluse/Desktop/Coding-Area/openwrt/designated_driver_trunk/feeds/luci/applications/luci-app-firewall/po/de/firewall.po /Users/bluse/Desktop/Coding-Area/openwrt/designated_driver_trunk/build_dir/target-powerpc_8540_musl-1.1.12/luci-app-firewall/ipk
g-all/luci-i18n-firewall-de/usr/lib/lua/luci/i18n/firewall.de.lmo;
bash: po2lmo: command not found
../../luci.mk:216: recipe for target '/Users/bluse/Desktop/Coding-Area/openwrt/designated_driver_trunk/bin/mpc85xx/packages/luci/luci-i18n-firewall-de_git-16.025.62197-445ae20-1_all.ipk' failed

Should po2lmo be added to luci.mk ?

Greetings Bluse

hnyman, 2016-01-26:
Works for me.
I just compiled trunk DE language module for Luci and e.g. the firewall module compiled normally:

    bin/ar71xx/packages/luci/luci-i18n-firewall-de_git-16.025.62197-445ae20-1_all.ipk

First guess: this is related to the recent $(STAGING_DIR_HOST) etc. changes in trunk and you have not built everything (including tools) from scratch in the past week. You might try "make dirclean" to completely clear the build, including the host tools. And then try the make again.

Or is there anything special in your build config or build host?

thuehn, 2016-01-26:
Thank you for the hint to dirlean before.
Now it builds on my side as well without probs.

closed 2016-01-26 by thuehn