openwrt/luci #645: Statistics' output config tab causes cbi dispatcher errors

hnyman, 2016-02-09:
Original report at https://dev.openwrt.org/ticket/21828 got me interested and I visited the Luci statistics setup / Output tab / network & rrdtool subtabs.
Original visit to "output tab" causes "firstchild" error, but both it and the later subtab selections then continue with cbi.map dispatches errors. Apply, Save etc. buttons are missing.

```
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/dispatcher.lua:449: Failed to execute firstchild dispatcher target for entry '/admin/statistics/collectd/output'.
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: The called action terminated with an exception:
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/dispatcher.lua:449: Failed to execute cbi dispatcher target for entry '/admin/statistics/collectd/output/network'.
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: The called action terminated with an exception:
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/nsection'.
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: A runtime error occured: [string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:11: attempt to concatenate field 'add' (a nil value)
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]: stack traceback:
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]:       [C]: in function 'assert'
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]:       /usr/lib/lua/luci/dispatcher.lua:449: in function 'dispatch'
Tue Feb  9 15:49:27 2016 daemon.err uhttpd[4220]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/dispatcher.lua:449: Failed to execute cbi dispatcher target for entry '/admin/statistics/collectd/output/rrdtool'.
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: The called action terminated with an exception:
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/nsection'.
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: A runtime error occured: [string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:11: attempt to concatenate field 'add' (a nil value)
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]: stack traceback:
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]:       [C]: in function 'assert'
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]:       /usr/lib/lua/luci/dispatcher.lua:449: in function 'dispatch'
Tue Feb  9 15:49:30 2016 daemon.err uhttpd[4220]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/dispatcher.lua:449: Failed to execute cbi dispatcher target for entry '/admin/statistics/collectd/output/network'.
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: The called action terminated with an exception:
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/nsection'.
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: A runtime error occured: [string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:11: attempt to concatenate field 'add' (a nil value)
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]: stack traceback:
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]:       [C]: in function 'assert'
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]:       /usr/lib/lua/luci/dispatcher.lua:449: in function 'dispatch'
Tue Feb  9 15:49:33 2016 daemon.err uhttpd[4220]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```
Error is visbile in LuCI (git-16.038.76221-cde9a62) / OpenWrt Designated Driver r48679 
but the output tab works ok in CC15.05: (git-16.034.35495-80a09b6) / OpenWrt Chaos Calmer 15.05-r48629 

Luci statistics has practically identical menu structure in trunk and CC15.05 (with only the openvpn and sensor items added in trunk), so I guess that this might be due to recent cbi.map changes and output tab's incompatibility with something. Possible?

jow-, 2016-02-09:
Should be fixed with cd85ef3

closed 2016-02-09 by jow-

hnyman, 2016-02-09:
Errors disappeared and the two problematic pages render ok after manually applying the patch to a live system.

Thanks.