openwrt/luci #646: Switch missing

jameskimmel, 2016-02-11:
I just freshly installed luci, but the switching section under network ist missing. 
Thanks for any help.
OpenWrt Chaos Calmer 15.05
LuCi git-15.276.51144-8832d53
Linksys WRT1900ACS
Let me know if you need further information. I'm totally new to OpenWrt/LuCI/Linux in general.

jow-, 2016-02-23:
Try the following on the console:

    uci add network switch
    uci set network.@switch[-1].name=switch0
    uci set network.@switch[-1].reset=1
    uci set network.@switch[-1].enable_vlan=1
    uci add network switch_vlan
    uci set network.@switch_vlan[-1].device=switch0
    uci set network.@switch_vlan[-1].vlan=1
    uci set network.@switch_vlan[-1].ports="0 1 2 3 4 5"
    uci commit network
    reboot

jameskimmel, 2016-02-29:
Thank you for your response. I will try that at the weekend. At the moment I already switched back to the original firmware.

cshoredaniel, 2016-12-10:
No additional comment for 10 months, closing.

closed 2016-12-10 by cshoredaniel