openwrt/luci #648: DNS configuration interface is missing

qiuzi, 2016-02-20:
Interface pictures:
<img https://drive.google.com/file/d/0B6RWOzqVsKcCd0Q1bzRlWXFnU2c/view?usp=sharing />

jow-, 2016-02-20:
Can you please be more specific what is missing exactly? From the picture it is not clear to me what you actually mean.

qiuzi, 2016-02-20:
Missing service providers api user and password
Error saving tips:
`
/usr/lib/lua/luci/dispatcher.lua:456: Failed to execute cbi dispatcher target for entry '/admin/services/ddns/detail/xxxxx'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/ddns/detail.lua:122: attempt to concatenate local '_source' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:456: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>`

chris5560, 2016-02-22:
What are the steps to get this "empty" picture ?
Could you please provide your /etc/config/ddns.

chris5560, 2016-02-22:
@jow
I need your support to fix this problem.

I found out the following:
- If a new ddns configuration is created than you get this "empty" screen quizi reported.
- If I comment out the dependency in line 324 of model/cbi/ddns/detail.lua which should read data from the ListValue use_ipv6 (defined in line 294) then all fields are shown.
- If you press save the you still get the error quizi reported. This value also depends on ListValue use_ipv6

For me it looks like that dependencies on ListValues are not handled correctly, because also other fields depending on Listvalue use_ipv6 are also not shown on other tabs.
Possibly also dependencies on MultiValues are also not working, but I don't know.

Thanks for your support.

closed 2016-02-23 by qiuzi