openwrt/luci #652: CBI Multi value dependent values not visible

Hostle, 2016-02-24:
The current refactoring of the cbi has broken the MultiValue dependency functionality. When the dependent value is enabled the dependents do not appear. Upon refreshing the page the dependants are visible but when cycled (disable/enable) they are no longer visibler. I have found a solution but I do not know  if it has any reverse effects ? However, perhaps it will help track down the issue.

/usr/lib/lua/luci/view/cbi/mvalue.htm
[code]
<%
    local i, key
    local v = self:valuelist(section) or {}
-%>

<%+cbi/valueheader%>
<% if self.widget == "select" then %>
    <select class="cbi-input-select" multiple="multiple" data-update="click change"<%=
        attr("id", cbid) ..
        attr("name", cbid) ..
        ifattr(self.size, "size")
    %>>
        <% for i, key in pairs(self.keylist) do -%>
            <option<%=
                attr("id", cbid.."-"..key) ..
                attr("value", key) ..
                attr("data-index", i) ..
                attr("data-depends", self:deplist2json(section, self.deplist[i])) ..
                ifattr(luci.util.contains(v, key), "selected", "selected")
            %>><%=pcdata(self.vallist[i])%></option>
        <%- end %>
    </select>
<% elseif self.widget == "checkbox" then %>
    <div>
        <% for i, key in pairs(self.keylist) do %>
            <label<%=
                attr("for", cbid.."-"..key) ..
                attr("data-index", i) ..
+              attr("data-depends", self:deplist2json(section, self.deplist[i]))..
+              ifattr(luci.util.contains(v, key), "checked", "checked")

            %>>
                <input class="cbi-input-checkbox" type="checkbox" data-update="click change"<%=
                    attr("name", cbid) ..
                    attr("value", key) ..
                    ifattr(luci.util.contains(v, key), "checked", "checked")
                %> />
                <%=pcdata(self.vallist[i])%>
            </label>
            <% if i == self.size then write('<br />') end %>
        <% end %>
    </div>
<% end %>
<%+cbi/valuefooter%>
[/code]

this fixes the issue of the dependents not being visible but I have also noticed on the  https://192.168.1.1/cgi-bin/luci/admin/network/iface_add page the "Create a bridge over multiple interfaces" has no effect and both the Flag and MultiValues are always visible. The above does not effect nor correct this could be an unrelated issue ?

jow-, 2016-02-24:
Hmm, I cannot really reproduce the problem, also `mvalue.htm` in current HEAD already has the `data-depends` line you're proposing, see https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/view/cbi/mvalue.htm#L29

The `iface_add` issue has been fixed with e78759ad5aa2418e7a39e1061d430c2e5e8fe0b8 and 55597da3a8540d19b34bf25ccf08d70e759013ab .

Can you retest with a current HEAD copy of `mvalue.htm` ?

Hostle, 2016-02-24:
I reverted my changes to the mvalue.htm and retested with your latest commits e78759a and 55597da and everything works as it should.

closed 2016-02-24 by Hostle