openwrt/luci #657: Luci client splash not working in offline

hi3520, 2016-02-29:
In luci client splash worked perfect in online it shows the splash page and redirect to predefined url .But in offline no splash page will appear and no redirect occurred. 
My project is just redirect every wifi user to a localhost eg:http://192.168.1.1**/welcome.html

FreifunkUFO, 2016-10-07:
that splash redirects http traffic to the splash page.
when you are offline, there will be no http traffic because dns-resolution will fail before any http traffic can start
so, how can we install/configure a dns faker?

cshoredaniel, 2016-12-10:
For clarification, is this for 'luci-app-splash' ?

LipkeGu, 2016-12-22:
@cshoredaniel: yes it is

LipkeGu, 2016-12-23:
@hi3520 What du you mean with offine?

FreifunkUFO, 2016-12-28:
such an offline dns could go like this:

echo "address=/#/192.168.1.1" > /tmp/dnsmasq.d/spoof2.conf; /etc/init.d/dnsmasq restart

thanks to http://serverfault.com/questions/351108/using-dnsmasq-to-resolve-all-hosts-to-the-same-address

cshoredaniel, 2017-01-22:
@hi3520 @LipkeGu Does @FreifunkUFO's solution work for you?

feckert, 2019-08-11:
Very old and I think the solution works because no reaction from issue creator

closed 2019-08-11 by feckert