openwrt/luci #660: luci-app-ntpc - broken working and visible error when 'option freq '0' to /etc/config/ntpclient

build000, 2016-03-02:
...as the topic.

cshoredaniel, 2016-12-10:
Probably ntpc is used very often since sysntpd takes care of what most folks would want.  Most likely if others confirm ntpc is not working in general (rather than specific options) it will get marked BROKEN.

build000, 2016-12-13:
ok - there is one simple solution, which previously simply did not notice - just enough there do not enter anything - then you can save changes to the luci without error - the question of whether this is the right way to configure NtpClient and whether it actually work? I would add that I use jesj only the bottom of any configuration and correct functionality dnscrypt-proxy.

cshoredaniel, 2017-01-22:
Seems like you've solved your problem, and it's more of a help issue for the forum or IRC.  Certainly obscure and needs more eyes than the issue tracker.

closed 2017-01-22 by cshoredaniel