openwrt/luci #668: cannot delete ssh keys

feckert, 2016-03-10:
Hello,

I cannot remove my added ssh key from the form section under administration.
Luci always complains that the form field is empty.
A workaround is to hint enter in the form and save this.

I have looked into the code under "model/cbi/admin_system/admin.lua". And add a validate function to check if the form is empty and if so return a new line Charakter '\n' but nothing happens the value will not be written to disk.

function keys.validate(self,value)
  local i=value
  if value == nil then
    i="\n"
  end
  return i
end

What can i do?

Kind Regards

Flo

hnyman, 2016-03-10:
Relevant code is found at https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/model/cbi/admin_system/admin.lua#L102

```
keys = s2:option(TextValue, "_data", "")
keys.wrap    = "off"
keys.rows    = 3
keys.rmempty = false

function keys.cfgvalue()
	return fs.readfile("/etc/dropbear/authorized_keys") or ""
end

function keys.write(self, section, value)
	if value then
		fs.writefile("/etc/dropbear/authorized_keys", value:gsub("\r\n", "\n"))
	end
end
```

The reason why nothing gets written with empty input is probably the if clause at line 113 in the write function.

I think that one alternative would be modify the if clause with an else statement to handle the empty value case. Something like
```
	if value then
		fs.writefile("/etc/dropbear/authorized_keys", value)
	else 
		fs.writefile("/etc/dropbear/authorized_keys", "")
	end
```

Or the if clause could be reformulated to first do the possible newline string substitution if there is value and alternatively set value to "\n" (or "", if that can passed to writefile). Something like 
```
	if value then
		value = value:gsub("\r\n", "\n")
	else 
		value = ""
	end
	fs.writefile("/etc/dropbear/authorized_keys", value)
```

feckert, 2016-03-10:
Hello Hannu,

i have already tested this. I added a sys.syslog("debug", "xxxx") message call into the function "keys.write" as you suggest. But the function will never called if the value is "nil". "nil" is the value which is passed into to the function if the form is empty.

Maybe you or some else could confirm my test results.

I am on the CC branch.

Regards
Flo

chris5560, 2016-03-11:
keys.rmempty = false

force that, if you want to remove all data, inside cbi AbstractValue.validate function the error "missing" is raised and write never called.

The main problem is inside cbi AbstractValue.parse function:
if rmempty = true and value = "" or nil, then user defined function write never called.
if rmempty = false and value = "" or nil then user defined function validate never called to return data, only to get user specific error message if there is one.

That's the reason why I used rewritten (value_)parse function as replacement inside luci-app-ddns, luci-app-radicale and luci-app-privoxy.
I not yet published that because there were a lot of changes inside cbi.lua the last weeks and month and for first testing inside self supported apps.

cshoredaniel, 2016-12-10:
Resolved in trunk, leaving bug open pending a decision regarding 15.05

feckert, 2016-12-13:
@cshoredaniel can i please show me the place where this issue was fixed in the trunk (master)?

cshoredaniel, 2016-12-21:
@feckert I was mistaken; that was specific to crontabs.  Sorry for the confusion (and I missed the notification of this message; sorry for taking so long).

closed 2018-06-25 by jow-