openwrt/luci #676: Cannot select stateless DHCPv6-Mode

diogocp, 2016-03-19:
I think the problem is in [modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/ifaces.lua](https://github.com/openwrt/luci/blob/e03f2b780a9ae28bdf270d36840b651fa723413e/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/ifaces.lua), lines [496](https://github.com/openwrt/luci/blob/e03f2b780a9ae28bdf270d36840b651fa723413e/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/ifaces.lua#L496) and [501](https://github.com/openwrt/luci/blob/e03f2b780a9ae28bdf270d36840b651fa723413e/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/ifaces.lua#L501). When selecting stateless, the option is removed from the config file, but then LuCI defaults to stateless + stateful.

hnyman, 2016-03-19:
Thanks for reporting.
As the default value for the option is 1, let's store explicitly "0" for stateless to avoid interpreting an empty/missing option to mean the default.

closed 2016-03-19 by hnyman

dlitz, 2016-04-03:
It's still not fixed.  When you disable the DHCPv6 server, luci deletes "dhcp.lan.dhcpv6" and "dhcp.lan.ra_management", so ra_management defaults back to 1 and odhcpd sends out router advertisements with the Managed bit set to 1.

See my notes here: https://dev.openwrt.org/ticket/16811

hnyman, 2016-04-04:
I copy my comment from the Openwrt bug discussion to also here:

After the fix here, setting the stateless-only mode  via Luci is now possible as long as you still leave the dhcpv6 mode option  as server.

For some reason, the stateless/stateful option (ra_management) in Luci has  been marked to be depending from the dhcpv6 mode being server or hybrid.  If dhcpv6 is set to disabled, that makes Luci to think that the  ra_management option is unnecessary and Luci deletes it, (which will then  lead into the default 1 (=stateless+stateful) being used.)

The dependency might be removed, but I am not quite sure if that is optimal. Any comments from @sbyx ? Would removing the dependency (of ra_management to dhcpv6=server/hybrid) from lines 500-501 be ok? Or could that lead into strange non-working configurations?
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/model/cbi/admin_network/ifaces.lua#L495

commit 355c2130 (2016-03-19): luci-mod-admin-full: Fix dhcpv6 ra_management option stateless value

Fix dhcpv6 ra_management option's handling of "stateless" value.
Default value for the option is 1, so trying to store ""
instead of "0" for stateless is not working.

This fixes #676
commit 2c34a421 (2016-03-19): luci-mod-admin-full: Fix dhcpv6 ra_management option stateless value

Fix dhcpv6 ra_management option's handling of "stateless" value.
Default value for the option is 1, so trying to store ""
instead of "0" for stateless is not working.

This fixes #676

(cherry picked from commit 355c21304afb0d388fb8bd6132e0dbcb19f03428)