openwrt/luci #677: collectd broken: bad argument #1 to 'pairs' (table expected, got nil)

ArianeLu, 2016-03-23:
**URL:**  /cgi-bin/luci/admin/statistics/graph/cpu
**Rev:**  r49067
**Error Message:**

/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute call dispatcher target for entry '/admin/statistics/graph/cpu'.
The called action terminated with an exception:
/usr/lib/lua/luci/statistics/datatree.lua:151: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

hnyman, 2016-04-16:
CPU graph works normally for me. Is anything special with your device?

ArianeLu, 2016-04-19:
Database store in external driver, file system is ext4 (mounted)

cshoredaniel, 2017-01-22:
I've been getting some funkiness with luci-app-statistics in some recent builds, but have not identified root cause (may be a case of need make distclean and a full rebuild).

hnyman, 2017-01-22:
> I've been getting some funkiness with luci-app-statistics in some recent builds, but have not identified root cause (may be a case of need make distclean and a full rebuild).

I have not noticed anything special. All 10 stats that I use, work ok.
(Conntrack, Processor, CPU Frequency, Entropy, Interfaces, Wireless, System Load, Memory, Ping, Thermal)

iv-mexx, 2017-01-22:
I experience this problem as soon as I change the storage directory to an nsf mounted directory. Data is written to the new directory, but I get this error in the web interface when I want to view graphs

hnyman, 2017-01-22:
So, common to both users with problem is that the data storage location is non-standard.

@iv-mexx
Could you share the relevant part of your luci-statistics config?
And does the error concern all stats plugins, or just some?

cshoredaniel, 2017-01-22:
@hnyman and that applies to my case as well; I've not been using default location.

iv-mexx, 2017-01-22:
This is what looks like the relevant part for me, not sure if thats all though:

```
config statistics 'collectd_rrdtool'
	option enable '1'
	option RRASingle '1'
	option StepSize '30'
	option RRATimespans '600 86400 604800 2678400 31622400'
	option DataDir '/mnt/nebula/logs/rrd'
```

Its happening for all the plugins I've active right now:

Interfaces: 

```
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute call dispatcher target for entry '/admin/statistics/graph/interface'.
The called action terminated with an exception:
/usr/lib/lua/luci/statistics/datatree.lua:151: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

Wireless: 
```
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute call dispatcher target for entry '/admin/statistics/graph/iwinfo'.
The called action terminated with an exception:
/usr/lib/lua/luci/statistics/datatree.lua:151: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

System Load: 
```
/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute call dispatcher target for entry '/admin/statistics/graph/load'.
The called action terminated with an exception:
/usr/lib/lua/luci/statistics/datatree.lua:151: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

cshoredaniel, 2017-01-22:
My config is similar except the RRATimespans uses the default '1hour 1day 1week 1month 1year', and I get a similar problem with the data dir as '/tmp/rrd' instead of /tmp/collectd/rrd'.

When I traced my error message (don't have handy at the moment) it seems uci:get and uci:get_all are returning nil instead data.  It's like luci.model.uci.cursor() can't find the config file any more (i.e. the directly thing suggests some sort of issue with path handling).  I'm actually wondering if the culprit is a recent ubox change (uci depends on ubox but uci hasn't changed recently, and this error has only shown up recently for me).

cshoredaniel, 2017-01-22:
Hm I noticed @ArianeLu's problem has been going on for a while, so may it actually was the uci update and just didn't get picked up in my case until I did distclean.

cshoredaniel, 2017-01-22:
I'm going to exclude my issue as a datapoint for now, since it's not clear if it's the same.

cshoredaniel, 2017-01-22:
Looking at the code in the above error messages, it appears in their case the data is not being found by the Instance._scan()

cshoredaniel, 2017-01-22:
@ArianeLu @iv-mexx What version of OpenWrt/LEDE & LuCI? r<blah> seem like it might be 15.05 or older.

iv-mexx, 2017-01-22:
I'm actually using Turris OS 3.5.1 which seems to be based on 15.05 of OpenWrt. 

> OpenWrt omnia 15.05 r47055 / LuCI 5ca9e5d2391f6ca149db4e53cda7c8f5d3ef6644 branch (git-16.335.29518-5ca9e5d)

ArianeLu, 2017-01-22:
@cshoredaniel Maybe early version of Designated Driver, It maybe resolved after reformat to FAT32, I'm not sure about this, just because it happened in 9 months ago, sorry.

cshoredaniel, 2017-01-22:
Ok, in @iv-mexx's case we'd need a test on stock 15.05 with corresponding luci (i.e. for-15.05 not trunk), and for @ArianeLu we're not sure it can be reproduced.  Therefore closing barring appearance of issue we can with something we develop.

cshoredaniel, 2017-01-22:
@iv-mexx I'd recommend first approaching Turris OS and if they find something from upstream (us) we can deal with their report.

closed 2017-01-22 by cshoredaniel

iv-mexx, 2017-01-23:
@cshoredaniel Ok, I just found this existing issue here fist (and no existing issue there)

jow-, 2017-01-23:
This error slso can happen if the relocated datadir has insufficient permissions. Remember that the disgram render pages run as user nobody, so ensure that your *.rrd files and all parent directories are world readable.

hnyman, 2017-01-23:
Thanks jow, that requirement is not that obvious to the users.
I added the advice to the LuCI GUI with eaafb4508

searene, 2019-06-19:
For anyone who is looking for a solution, here is one. `ssh` to your router and execute the following command.

```
rm -r /tmp/luci-indexcache
```