openwrt/luci #696: Missing `Etc` timezones

patricktokeeffe, 2016-04-05:
Time zone list is missing the `Etc/GMT±X` POSIX zones usually found on Linux devices. Users who require ignorance of daylight savings times are force to use UTC, which may not be an acceptable alternative. 

Semi-related: #675

hnyman, 2016-04-05:
The script https://github.com/openwrt/luci/blob/master/build/zoneinfo2lua.pl uses a Linux system's /usr/share/zoneinfo/zone.tab as the source for the zone list. That list comes from the IANA zoneinfo files, but is "country-based" and does not include the Etc timezones like GMT+-X and thus their POSIX strings are not parsed into Luci's tzdata.lua file. So there is no Luci support for those zones.

Note: You would not benefit from the GMT+-X zones, as their POSIX zone string looks like ``` <GMT+1>1 ```, which musl currently mishandles, as I noticed in #675 

But after looking at the musl implementation ( http://git.musl-libc.org/cgit/musl/tree/src/time/__tz.c#n118 ), I think that you might succeed from console if you install the full "zoneinfo" package into your router and manually select the timezone in /etc/config/system (as ``` <GMT+1>1 ```). I think that then linux and musl will use the correct info from the zoneinfo files (instead of just the POSIX string that fails in musl). I did not test that, 

Alternatively, you could just invent a private timezone and edit its string into /etc/config/system. Musl works ok as long there are only characters in the zone name. (I edited /tmp/TZ directly to be PPP-4 in the next example, but that would not survive reboot)
```
root@OpenWrt:~# date -u
Tue Apr  5 07:31:22 UTC 2016

root@OpenWrt:~# cat /tmp/TZ
EET-2EEST,M3.5.0/3,M10.5.0/4
root@OpenWrt:~# date
Tue Apr  5 10:31:31 EEST 2016

root@OpenWrt:~# nano /tmp/TZ
root@OpenWrt:~# cat /tmp/TZ
PPP-4
root@OpenWrt:~# date
Tue Apr  5 11:31:58 PPP 2016

root@OpenWrt:~# grep " r" /etc/banner
 DESIGNATED DRIVER (Bleeding Edge, r49114)
```

hnyman, 2017-05-08:
Fixed in LuCI master with https://github.com/openwrt/luci/commit/f8eee09f55faa375b11cf00b2f96915c92f924f4 and https://github.com/openwrt/luci/commit/f2cf516cf7d50c191e3ea94f4696a6c60348cf92

closed 2017-05-08 by hnyman

patricktokeeffe, 2018-01-05:
..a belated thank you!