openwrt/luci #697: Fail to open Luci OpenVPN

dafresh90, 2016-04-07:
hi guys, 
openvpn worked fine for me with 15.05 CC. I've updated to 15.05.1 and now the OpenVPN doesn't work. The config was kept after upgrading but the software for OpenSSL and DDNS needed to be installed again. DDNS seems to be working fine, but the OpenVPN GUI does not work with following message:

/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/services/openvpn'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:311: Unable to read UCI data: openvpn
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

thanks in advance ;)

hnyman, 2016-04-08:
> The config was kept after upgrading but the software for OpenSSL and DDNS needed to be installed again. DDNS seems to be working fine, but the OpenVPN GUI does not work with following message:

Have you installed the openvpn package itself?
(You mention reinstalling just opsnssl and ddns, but not openvpn)
If not, you probably need openvpn-openssl, as you are apparently using the openssl variant of it.

And yes, you need to reinstall all needed software modules after the firmware upgrade. Sysupgrade only preserves the configs, not the binary programs & libraries.

dafresh90, 2016-04-08:
Hi hnyman, 
thanks for your quick reply. 
I reinstalled the openvpn-luci-app, openvpn-openssl, easy-rsa and ddns. The configfiles seem to be fine like before. DDNS work like a charm and OpenVPN log reveals "sequence complete" but connection could not be established.

I recovered the backup file for 15.05 and 15.05.1 but still the same error. Seems like the other features work well.

Wedmer, 2016-04-19:
check read-write permissions

Wedmer, 2016-05-10:
Looks like http://luci.subsignal.org/trac/ticket/344

cshoredaniel, 2017-01-17:
No reponse for months, closing

closed 2017-01-17 by cshoredaniel