openwrt/luci #701: Address the lack of visibility in the private network devices WIFI type phone with android.

build000, 2016-04-16:
As is well known in the latest compilation Trunk is now seen more detailed list of the client connection WIFI connected to our AP. It is with this little problem, poniważ in a typical configuration with an android phone, the name of the device consists of the word "android-" and the following individual, random markings in the form of 16 digits - the effect on the "Status" (/ cgi-bin / luci / admin) is that it is never seen the full address of the LAN such client WIFI, which is assigned to our AP.

http://i.imgur.com/hAJyEbl.png

hnyman, 2016-04-16:
Based on your picture, that is likely due to the browser being smart and selecting "-" to place the necessary line break.

It might be better to forcufully divide the info onto two lines, device name and IP, as the TX/RX info is on two lines in any case.

build000, 2016-04-16:
A proposal:
Let these problematic data is always displayed in at least two rows:
1 row - the name of the device with a sufficiently large number of displayable characters
2nd row - only the IP address on the LAN

closed 2023-10-06 by build000