openwrt/luci #708: adblock plugin - problem global saving changes

build000, 2016-05-01:
Any update (eq. when restart router or...when enable cron service to update) not automaticaly saving changes = you have to manually enter changes each time.

hnyman, 2016-05-01:
I am not quite sure what you do mean. the luci-app-adblock plugin just uses normal uci to store settings like any other change made via luci.

@dibdot

build000, 2016-05-01:
So I have added more information about this error in "packages".

dibdot, 2016-05-01:
I can't see any problem. You've enabled all block list at once and adblock only commits all changes at the end of the processing to config (to mimize flash updates). So please wait until adblock processing has been finished - same applies to start/stop/restart cycles.

build000, 2016-05-01:
Also just I thought - so much that it took 25 minutes and no change - luci still requires manual saving changes ...

dibdot, 2016-05-01:
in your screenshot I'll see a processing time of about 2 minutes for all lists ... did you see a final log entry like ...
`adblock[8353] info : domain adblock processing finished successfully (1.1.0, r49276, 01.05.2016 20:52:31)`
? If not, please disable a big list like shalla and try it again ... maybe a resource limit on your router.

build000, 2016-05-01:
I'll be back for a moment to what I wrote early in this and another post:

Seemingly while the error no matter what ideology we adopt in determining the issue of "readability interface luci", is that zwyczjnie disappears position service adblock list on the page / tab management available in the luci services at a time when for once we use on this side of the "few "start / stop / restart.

dibdot, 2016-05-01:
Please answer my question! 
adblock 1.1 adds 3 new block list sources ... and do the config update/commit **only** at the end of processing. Use default setup (in /etc/adblock/adblock.config.default with 3 enabled lists) and start from there ... still it doesn't make sense to always restart an already running service.

build000, 2016-05-01:
So the fundamental question - why so plugin luci (or /etc/config/adblock) as many possible letter to the inclusion, since their inclusion causes a problem with the operation of both the script and the luci ???

build000, 2016-05-04:
important observation: in the last version adblock with a large number of addresses (more loaded lists than the default) never appears in the log line "finished" - [b]here's the problem[/b] - because luci always shows the need to manually save your changes, in spite of that (I presume ) addresses are loaded into the firewall ...

dibdot, 2016-05-04:
Did you re-test with the latest version (1.1.1)? Did it work with the default setup? If so, please drop me a mail to my maintainer email address (see readme) and I'll send you a debug version.

build000, 2016-05-05:
Yes - if use "default" settings (not any config), its ok, like this:
`Fri May  6 00:46:53 2016 user.notice adblock[1999] info : domain adblock processing finished successfully (1.1.1, r49296, 06.05.2016 00:46:53)`

dibdot, 2016-05-06:
It's probably unrelated to luci and unrelated to your multiwan setup, I think it's a memory (hardware) issue on your router. The adblock process dies silently after memory intense sort processing and after that your system seems to be "borked" - until the next reboot.
As I wrote before, to nail this down write to my maintainers email address ...

build000, 2016-05-06:
And it seems to me that this was recently made changes in the script make this probelm - with version 1.0.1 I set up a cron restart service at 2-ie hours and the process always ended properly line "finished" in the logs ...

closed 2016-05-09 by build000