openwrt/luci #713: Headings in bootstrap theme do not support hrefs if they have grandchildren

sarumpaet, 2016-05-06:
...see https://github.com/openwrt/luci/blob/master/themes/luci-theme-bootstrap/luasrc/view/themes/bootstrap/header.htm#L119 , but there are several packages that seem to rely on this (luci-mod-freifunk's "Status" page, for example, is linked to directly in the heading).

cshoredaniel, 2016-12-10:
@sarumpaet Which version of LuCI/OpenWrt, please

cshoredaniel, 2017-01-17:
Actually material doesn't support href's in headers either; I'm not sure if this should be classed as a theme bug or if it's been more of a use of non-promised functionality that just happened to work and now does and the bug really is in the the apps, etc.

feckert, 2019-09-18:
This is a very old issue.

closed 2019-09-18 by feckert