openwrt/luci #720: luci.model.ipkg.status/lookup doesn't handle opkg failures

karlp, 2016-05-11:
As mentioned in the comments, opkg sucks.  I had been executing ```opkg status <pkg-name>``` directly myself, but after getting errors like 
```
* opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
```
I was trying to use luci.model.ipkg.status(xxx) instead, and when calling this in rapid succession for numerous packages, I was, now simply getting a returned data block that was empty.

This is because the "os.execute("opkg status"..." line doesn't check the return code, and throws away stderr.  See https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/model/ipkg.lua#L87

A potential improvement would be...

```Lua
        local rc = os.execute(cmd .. (" >%s 2>>/tmp/opkg.fails.lua.log" % tmpfile))                                                                               
        local data                                                                 
        if rc == 0 then                                                            
                data = _parselist(io.lines(tmpfile))                                 
        else                                                             
                data = nil                                               
        end                                                              
                                                                         
        os.remove(tmpfile)                                               
        return data                   
```

But that doesn't really help fix the problem that opkg is just sucking on this.  What I've done instead is get opkg status for the entire system, and parse that myself for the pkgs I care about.

Footnote, what on earth is the difference between opkg status and opkg info?!

cshoredaniel, 2016-12-10:
@karlp I seem to recall discussions of this and some resolution, but don't know where I saw that discussion.  Am I misremember, or can this bug be closed?

karlp, 2016-12-12:
My resolution internally was to not use status/lookup for single packages, but to fetch the entire list of packages and scan for the ones I was interested in.  I'm not aware of any resolution to the issue as filed.

cshoredaniel, 2017-01-17:
@karlp first let's clarify; is there a solution without changing opkg?  If not then an opkg fix to core needs to be made first.

karlp, 2017-01-17:
Allow me to requote:
```
This is because the "os.execute("opkg status"..." line doesn't check the return code, and throws away stderr.
```

That's in the luci side.  So yes, luci side could absolutely handle the failures properly, which is why I filed the bug.

cshoredaniel, 2017-01-17:
You also said `But that doesn't really help fix the problem that opkg is just sucking on this. What I've done instead is get opkg status for the entire system, and parse that myself for the pkgs I care about.` which leads me to wonder whether you are saying dealing with the return code and stderr is not enough.

karlp, 2017-01-17:
Yes, I worked around it by making a single call to opkg where I know I should always get results, so i can check it manually that way.  The thing is, luci has this method, "luci.model.ipkg.status" which, if called repeatedly, will start giving erroneous results. (You'll get an empty set back, which is also the _correct_ response for a  non-installed package!)  The _reason_ it gives erroneous results is because opkg is sucky.  But luci should be able to handle this more gracefully, which is what _this_ bug is about.

cshoredaniel, 2017-01-17:
Ok, got it.

jow-, 2021-12-03:
This class is deprecated and not used by LuCI itself anymore.

closed 2021-12-03 by jow-

karlp, 2021-12-06:
My workaround of fetching the entire list is still in place, but you're deprecating the entire class right?  Are the alternatives: a) execute opkg directly, and b) convert to client side via... ? There's nothing obvious under https://openwrt.github.io/luci/jsapi/index.html, i presume it's declaring some effectively undocumented ubus method?

buana-bbl, 2024-05-16:
I have the same issue for "* opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable."...
but  already [SOLVED] by moving the "dest ram" ( in /etc/opkg.conf ) to a folder/directory in another bigger size partition.
its works for now
I hope this will help...