openwrt/luci #724: Status-Overview missing host information

danlight, 2016-05-15:
I'm using OpenWRT on a couple of Meraki MR12's. On "Status-Overview" the list of associated stations show only the MAC-address. It would be convenient to be able to configure what to show, with MAC always shown and IP and DNS-name selectable.
This applies also to "Network-Wifi", which already shows the IP-address.
Or simply print MAC + DNS and if DNS is not available MAC+IP and if IP is not available (AP/Router did not provide IP and don't have the IP-address in the ARP-table yet) lookup from DNS or just show the MAC-address.
Thank you!

danlight, 2016-05-15:
I should have said that on the Status-Overview page the host column shows a ? on the MR12

cshoredaniel, 2016-12-10:
To clarify, you are using the devices as AP's with a bridge to the network with a DHCP server or whatnot?  So the devices don't show IP because the device hasn't seen the IP only the MAC, so far (or else would have to correlate with the ARP table and/or DNS)?
This would definitely be useful, but is a case of 'patches welcome', I think.

cshoredaniel, 2016-12-10:
@danlight Actually this is a duplicate of #540 which has a lot of activity, and appears to be getting some traction.

cshoredaniel, 2017-01-22:
Actually this is the only part of #540 left and for cleaner comment history the relevant comments should be moved here.

th0ma7, 2017-01-23:
Interesting as this is exactly the issue I have :)
I thought that auto-magically if a hostname is not found (within luci wireless associated stations) that it would use "dig" or similar to find the corresponding hostname using the DNS forwarding IPs (if set)... if not found, print the IP.

braian87b, 2017-05-16:
A simple nslookup call inside any AP (a query to dnsmasq that is giving DHCP leases on the MainRouter) should bring the name from the MainRouter running from any AP...

It is just so simple but I don't know yet why can't this be added as optional feature to fill a file somewhere, and read data from there and show it on luci:

https://github.com/openwrt/luci/issues/540#issuecomment-244658651

see that post only (ignore the others testings between me and juanriccio.

th0ma7, 2019-11-14:
I might have found a relatively ugly hack to make things working...

All commands below are to be executed from the Acess Point (AP).  Basically the idea is to ensure your AP are ssh'trusted to your main router where the DHCP server is running so it's possible to `scp` every 15 minutes the dhcp leases file from, assuming all are running OpenWRT.

Create a `ssh` private+public dropbear key:
```
root@openwrt:~# dropbearkey -t rsa -f ~/.ssh/id_dropbear
Generating 2048 bit rsa key, this may take a while...
Public key portion is:
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQCEn1FWEJvF1MamXSDoB82OKDFvVSAVjYdMpRQuXLhqVZR3y9zlQDhIuATe9OxkE81i7vBgFygD5QYJZ2HsauLTkQlGG9BDFLtxMPpNxM6y7XYazgWVssa+VZeUpaBPLkBz7lvD1y0qB92QlLcxQf6YIZfFT/Pjb3GuF7G3PLtG/09cM5Qh4D09bF/1vdPNFp1Ez6uk5UebfrTYoir3dFSx0E53gUIUicvYwUrP3t18znnKKqi/hh2Tq5YnCgYnJk9fSbcH6vEt/XSN2CZ3oaADm1uGT240ZPyWS+XPInmO3qbGyOQlUN45oNbINJNAMedTqfG/KRl4pHN8qSKKDxNR root@my-ap-wifi
Fingerprint: sha1!! 90:34:d7:63:b0:db:4c:1e:23:9f:77:c5:f5:13:b7:ca:b8:e2:9d:0b
```
Extract the public key to a file:
```
root@openwrt:~# dropbearkey -y -f ~/.ssh/id_dropbear | grep "^ssh-rsa " >> .ssh/id_dropbear.pub
```
Copy AP root account public key to your main router so it can be trusted:
```
root@openwrt:~# cat ~/.ssh/id_dropbear.pub | ssh root@192.168.80.1 "cat >> /etc/dropbear/authorized_keys && chmod 0600 /etc/dropbear/authorized_keys && chmod 0700 /etc/dropbear"
```
Create a `cron` job and restart the cron daemon:
```
root@openwrt:~# echo "0/15 * * * * scp root@192.168.80.1:/tmp/dhcp.leases /tmp" > /etc/crontabs/root
root@openwrt:~# chmod 600 /etc/crontabs/root
root@openwrt:~# /etc/init.d/cron restart
```
You can use `logread` to watch for `cron` daemon output such as:
```
root@openwrt:~# logread -f
Wed Nov 13 21:39:00 2019 cron.info crond[12505]: USER root pid 12515 cmd scp root@192.168.80.1:/tmp/dhcp.leases /tmp
```
And confirm file is being updated in `/tmp`:
```
# ls -1la /tmp/dhcp.leases 
-rw-r--r--    1 root     root          1755 Nov 13 21:45 /tmp/dhcp.leases
```

And voilà!  The main status page should now show associated stations names...

jow-, 2019-11-14:
LuCI master will now add DNS information in case it is available. Note that not all cases can be resolved, e.g. when the 802.11 BSSID in use does not correspond to any actual ethernet MAC used for IP communication.

closed 2019-11-14 by jow-

th0ma7, 2019-11-16:
Can this be backported to 19.07 as bug-fix?