openwrt/luci #725: Broken LuCi after installing luci-app-statistic

djesys, 2016-05-20:
instead of webinterface I get:

`.../luci/controller/luci_statistics/luci_statistics.lua:11: module 'luci.statistics.datatree' not found:
	no field package.preload['luci.statistics.datatree']
	no file './luci/statistics/datatree.lua'
	no file '/usr/share/lua/luci/statistics/datatree.lua'
	no file '/usr/share/lua/luci/statistics/datatree/init.lua'
	no file '/usr/lib/lua/luci/statistics/datatree.lua'
	no file '/usr/lib/lua/luci/statistics/datatree/init.lua'
	no file './luci/statistics/datatree.so'
	no file '/usr/lib/lua/luci/statistics/datatree.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	.../luci/controller/luci_statistics/luci_statistics.lua:11: in function 'v'
	/usr/lib/lua/luci/dispatcher.lua:536: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:201: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function `</usr/lib/lua/luci/dispatcher.lua:167>``

I tried rm /tmp/luci-indexcache and reboot, but with no success. I even tried opkg remove luci, but it doesnt do anything. 

I thought maybe reinstalling the luci-app-statistic package will work but I get:

`root@OpenWrt:~# opkg install luci-app-statistics
Installing luci-app-statistics (git-15.248.30277-3836b45-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/luci/luci-app-statistics_git-15.248.30277-3836b45-1_all.ipk.
Installing collectd (5.4.2-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/packages/collectd_5.4.2-1_ar71xx.ipk.
Collected errors:
 * copy_file: unable to open `/etc/collectd.conf-opkg.backup': Read-only file system.
 * file_copy: Failed to copy file /etc/collectd.conf to /etc/collectd.conf-opkg.backup.
 * backup_make_backup: Failed to copy /etc/collectd.conf to /etc/collectd.conf-opkg.backup
 * opkg_install_cmd: Cannot install package `luci-app-statistics.``

Please help

hnyman, 2016-05-26:
Based on those error messages I guess that you have problems with the file system "read-only file system". Likely your disk is full, or something like that.

In general, you should update to 15.05.1

LipkeGu, 2016-12-22:
`I tried rm /tmp/luci-indexcache and reboot, but with no success. I even tried opkg remove luci, but it doesnt do anything.`

removing files on jffs2 doesnt free storage... full is full

closed 2016-12-22 by hnyman