openwrt/luci #726: LuCI-App-samba - Get Rid of "smb.conf.template" =(

LipkeGu, 2016-05-22:
Since Years Buildroot and Freetz also OpenWRT uses an HUGE textbox to configure samba.conf and should be more like easier maintainable :( sometime i had seen the template is very vuggy configured, because of non-understanding what these Options do... so we should "Facelift and cleanup the Form and settings pages instead of using textoutput in a textbox. LucI can more such printing textboxes :(

cshoredaniel, 2016-12-09:
Unfortunately this is likely not to happen unless a community member creates a suitable PR.

LipkeGu, 2018-02-14:
I mean something like this...

[image]