openwrt/luci #731: don't use 301 permanent redirect

basinilya, 2016-05-27:
The web-interface redirects from the root (​http://192.168.1.1) to the login page (​http://192.168.1.1/cgi-bin/luci) with a 301 Permanent redirect. This has bad implications in some browsers (I noticed with Chrome), since they cache 301s. The better choice would be a 302 temporary redirect.

When I went a other network and connected to a different router the browser kept redirecting me.

LipkeGu, 2016-05-27:
Firefox has also such behaver :/ :+1:

jow-, 2016-05-28:
The thing is that there is no explicit 301 redirect I could change, it is a static html page with a meta refresh tag.

We could try to add pragma nocache and expiry tags but not sure if browsers would obey them.

basinilya, 2016-05-28:
I see a new line:

    <meta http-equiv="Cache-Control" content="no-cache" />

added a few days ago. I guess it's not enough, because it's not widely supported.

LipkeGu, 2016-05-31:
@jow- you could change the valid-time for this document :)

jow-, 2016-06-08:
@LipkeGu - how exactly?

J0WI, 2017-08-17:
I can confirm this issue. It's very bad if you want to flash multiple devices with an open firmware or if you switching between different networks.
The redirect prevents you from accessing the login interface of new routers without luci.

closed 2017-08-17 by jow-

J0WI, 2017-08-17:
@jow- could you please add a statement why you just closed this issue?

jow-, 2017-08-17:
There is no 301 redirect emitted by the server, merely a static html page with a meta refresh tag and a nocache pragma tag.

J0WI, 2017-08-17:
However browsers seems to cache this. Can we keep this issue open or create a new one to find a workaround?

jow-, 2017-08-17:
For what purpose? No movement in over a year. If at all the issue needs to be fixed in uhttpd which is not part of LuCI.

J0WI, 2017-08-17:
Some links for further reading:
* (Old) ticket on OpenWRT: https://dev.openwrt.org/ticket/22105
* (Old) status page of Gluon: https://github.com/freifunk-gluon/gluon/blob/2d8dda6eeeb5557def23e91ed712c7f5b2ae0763/package/gluon-status-page/files/lib/gluon/status-page/www/index.html

klunejko, 2017-10-01:
Ah, so this is what the problem was. Has been annoying as hell, having to clear the browser cache time.

In any case, that single meta tag doesn't work, hence why browsers cache the page. Ideally you would simply set the expires/cache headers via httpd. If you insist on using html only, the following will/should prevent caching by the major browsers.

`<meta http-equiv="cache-control" content="max-age=0" />`
`<meta http-equiv="cache-control" content="no-cache" />`
`<meta http-equiv="expires" content="0" />`
`<meta http-equiv="expires" content="Tue, 01 Jan 1980 1:00:00 GMT" />`
`<meta http-equiv="pragma" content="no-cache" />`

And all those meta tags are needed, as each browser interprets meta tags differently or users might be using proxies, older http version or what have you.

For html5 you would need to add a cache manifest.

`<!DOCTYPE html>`
`<html manifest="manifest.appcache">`

`CACHE MANIFEST`
`# Cache manifest version 1.0`
`# no cache`
`NETWORK:`
`*`

fildunsky, 2024-12-28:
Seems almost 10 years has passed but still problem exists. If I flash lot's of different firmwares to my OpenWRT router and then change IP from 192.168.1.1 to anything else, Chrome saves 192.168.1.1 cache and I can't open my other, main non OpenWRT router admin page since it starts redirecting to /cgi-bin/luci/ from cache.