openwrt/luci #734: LUCI crash after running in 15.05.01 release

bilik, 2016-05-31:
Router model: TP-Link TL-WR1043ND. 
After change uhttpd => lighttpd.
Every time I use LUCI crash with mesage.
thanks in advance
Everything another is OK (php, html, scripts, rrd, ...)

/usr/lib/lua/luci/util.lua:587: Unable to establish ubus connection
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/util.lua:587: in function 'ubus'
	/usr/lib/lua/luci/dispatcher.lua:321: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

bilik, 2016-05-31:
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/util.lua:587: Unable to establish ubus connection
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

jow-, 2016-05-31:
Make sure luci runs as root.

bilik, 2016-06-01:
But I can not run web server (lighttpd) as root. Is any another choice?

Rom4nB, 2016-07-08:
I have the same issue with the same router.

Is there anything new about this?

jow-, 2016-07-08:
What same issue? That LuCI is not running as root?

Rom4nB, 2016-07-08:
No, that luci won't start but show the message
/usr/lib/lua/luci/util.lua:587: Unable to establish ubus connection

I tried installing OwnCloud using this howto
https://wiki.openwrt.org/doc/howto/owncloud

It works fine until the part with "disable httpd and use luci on lighttpd". I could not manage to get luci working again, I only got the same error message as blllk above.
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/util.lua:587: Unable to establish ubus connection

Finally I used 
firstboot

to regain the default settings and luci works again.

hurenkam, 2016-07-11:
The following (workaround) fixes this issue for me on lighttpd:
root@OpenWrt:~# chmod +s /usr/bin/lua

Warning: This is probably not a safe solution if you care about security, especially if you are using lua for other stuff.

cshoredaniel, 2016-12-10:
@bilik @Rom4nB @hurenkam I'd recommend changing the listening port for either lighttpd or for uhttpd and running the both - uhttpd for the web UI and lighttpd for you actual web server.

cshoredaniel, 2017-01-17:
This is unlikely to solved in current generation of LuCI; luci-ng has aspirations to use better techniques.

cshoredaniel, 2017-01-22:
Acutally I'm reopening this issue so hopefully other can find it more easily.

gstrauss, 2017-02-02:
More details, please.  lighttpd can be run as root, if that is what is desired, and lighttpd works fine with LuCI in Turris Omnia, which runs software based on openwrt, and does not run uhttpd.

cshoredaniel, 2017-02-04:
@gstrauss It looks to me like the users above are complaining that they want LuCI to work without it running either as root or making lua setuid root.  As you can imagine that isn't an option with current LuCI.

gstrauss, 2017-03-02:
There are multiple ways to run backend scripts under a different user id than the web server.  I believe that those running lighttpd in order to run other hosting, scripts, and frameworks do not want lighttpd to run as root (and all those scripts as well) just so that LuCI can run as root.  (FYI: lighttpd will run as the user that started lighttpd if `server.username` and `server.groupname` are omitted from lighttpd.conf or commented out.)

A quick workaround to _reduce_ what is running as root would be to rename cgi-bin/luci to cgi-bin/luci.privileged and to create a setuid root program cgi-bin/luci which turned around and executed /full/path/to/cgi-bin/luci.privileged.  That is still not ideal and nobody other than root should be able to write to that cgi-bin directory, but it is a **much better** solution than setting chmod g+s on /usr/bin/lua (suggested by @hurenkam), which I would never, ever consider doing.

gstrauss, 2017-03-02:
I think this issue can be closed.

Here's the tl;dr
* If you want to run LuCI with lighttpd, then lighttpd can be configured to run as root by commenting out `server.username` and `server.groupname` in /etc/lighttpd.conf.  If you are **only** running lighttpd for access to LuCI, then this is an "ok" solution.  Alternatively, keep uhttpd running (as root).
* If you are using lighttpd to run owncloud or any other framework, then a more secure solution is to run multiple instances of lighttpd (on different ports), or multiple web servers (on different ports), with **one** instance running as root and serving LuCI, and the other (non-root) instance running everything else.
* If you want to run a single web server to do all of the above, then consider running lighttpd as non-root, and having a single, simple setuid root executable whose only function is to execute luci, as mentioned in the previous post.

cshoredaniel, 2017-03-03:
@gstrauss Thanks for the clarification; I didn't actually mean that you had to run lua setuid root but that 'something has to run suid root', which you've explained well for those who want to do such things; even then I really look forward to time to working on luci-ng and letting ubus/rpcd ACLs manage privileges and to need a webserver on root; however that is quite a way out unless something changes in either available time and/or priorities.  I will however leave the bug, open, but with appropriate tags, so that its easier to find, unless @jow- or @hnyman think it should be closed.

closed 2018-07-03 by jow-