openwrt/luci #736: luci-ssl is broken with the latest chrome

neheb, 2016-06-02:
works fine with firefox. it just times out at establishing secure connection. no idea why.

i remember there was a news story on how google chrome disabled npn which broke some sites. no idea if luci-ssl also uses npn.

Vulpecula-nl, 2016-06-09:
I have the same problem. Its not working anymore in google chrome version 51.0.2704.84 m (64-bit). I have changed the parameter redirect_https to 0 in /etc/config/uhttpd and now it works again, but not with ssl connection

hnyman, 2016-06-09:
I noticed the same when trying to connect with my Android tablet. No more Luci :-(

jow-, 2016-06-09:
Cannot do much about that. Sounds like something that needs to be addressed in the various crypto backends (ustream-ssl, polar/wolfssl, cyassl/mbedtls, openssl).

jow-, 2016-06-09:
Confirmed, swapping libustream-polarssl with libustream-cyassl works, so neither a LuCI, nor a uhttpd issue. Likely needs a library update to polarssl/mbedtls.

hnyman, 2016-06-09:
I tested switching from polarssl to its new incarnation mbedtls, and that works with Chrome, too.
So the culprit it is just the old/ancient polarssl. 

Would it be time to switch the default from polarssl to mbedtls?

blogic, 2016-06-09:
whats the size difference ?

hnyman, 2016-06-09:
Roughly:  polarssl 141 kB, cyassl 147 kB, mdebtls 162 kB, openssl 735 kB

from ar71xx:
```
libcyassl_3.9.0-1_mips_34kc.ipk                    08-Jun-2016 07:46              147593
libmbedtls_2.2.1-1_mips_34kc.ipk                   08-Jun-2016 07:33              162085
libopenssl_1.0.2h-1_mips_34kc.ipk                  08-Jun-2016 07:49              735634
libpolarssl_1.3.16-1_mips_34kc.ipk                 08-Jun-2016 07:33              141441

libustream-cyassl_2016-06-07-17085b7abc0cd09c64..> 08-Jun-2016 08:07                3842
libustream-mbedtls_2016-06-07-17085b7abc0cd09c6..> 08-Jun-2016 08:07                4336
libustream-openssl_2016-06-07-17085b7abc0cd09c6..> 08-Jun-2016 08:07                4146
libustream-polarssl_2016-06-07-17085b7abc0cd09c..> 08-Jun-2016 08:07                4233
```

hnyman, 2016-06-09:
Note that switching from polarssl to cyassl or mbedtls would have impact on a few non-core packages, as they do not have variants / config options for non-polarssl. (However, most apps using polarssl like transmission, umurmur, openvpn and shairport have also openssl versions, so they have a backup option.)

px5g, the creator of the self-signed certificate for Luci / uhttpd, requires polarssl at the moment. That is probably the item on the critical path to change away from polarssl. As long as px5g only supports polarssl, there is no practical way to switch away from it.

jow-, 2016-06-11:
I believe the issuse is solved with https://github.com/lede-project/source/commit/9e45f9d63c797baab2b3bd2574990ee94d3f80cb

neheb, 2016-06-15:
Issue is fixed. That's really bizarre that GCM is required now. Maybe google is more aggressive with TLS 1.2? A different solution would probably be to disable TLS 1.2 and keep 1.1. GCM is not available with 1.1. I'm not sure if size is a real concern here.

closed 2016-06-15 by neheb

rvalles, 2016-06-15:
Fixed in lede, but not openwrt?

dircleaned, pulled, built and installed designated driver right now, issue still present.

Dekker500, 2016-06-22:
OpenWrt Designated Driver snapshot r49388 from June 21 still exhibits the symptoms.