openwrt/luci #738: glitch with luci

JazzMaster, 2016-06-05:
chmod -R +x /www/cgi-bin
is needed once installed. otherwise it works fine.

as an adblock trick, screw pixelserve, and use uhttpd itself. make a basic 404 page and put this into it:
(link, see bottom of page) http://png-pixel.com/

it doesnt seem to log, and im not sure how to turn logging off anyhoo. but it works.

jow-, 2016-06-05:
@JazzMaster - I believe the chmod issue is due to faulty OpenWrt buildbots. They run with a bad umask setting causing every Nth LuCI build to be shipped with wrong permissions.

jow-, 2016-06-28:
I cannot reproduce the problem locally or on LEDE project builds. If you see another snapshot build with bad /www permissions then please notify the OpenWrt developers that one of the build slaves has a bad umask setting.

closed 2016-06-28 by jow-