openwrt/luci #742: luci-app-statistics:

runeevjen, 2016-06-15:
Hello,

I manually applied the changes in [3289e134953887482bdaa28c28567917bb91facb](https://github.com/openwrt/luci/commit/3289e134953887482bdaa28c28567917bb91facb) (4 changed files) to LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55) / OpenWrt Chaos Calmer 15.05.1.

I discovered two issues:
**Issue 1:** In Luci Statistics->Setup->Network plugins->OpenVPN only the following text is shown:

> OpenVPN Plugin Configuration
> The OpenVPN plugin gathers information about the current vpn connection status.

I.e. no selections (enable this plugin etc) are shown in luci

After adding config statistics 'collectd_openvpn' with settings in /etc/config/luci_statistics the OpenVPN settings worked, and changed settings in luci were reflected in /etc/config/luci_statistics 

**Issue 2:** After having a working configuration OpenVPN menu item is not shown in  Luci Statistics, and RRD does not generate files related to openvpn in /var/rrd/<hostname>

I have restarted the services and the router, and all other collectd plugins are working as expected,

Best regards,

Rune Evjen

hnyman, 2016-06-16:
If the data file is not generated, then the problem is likely on the collectd side (packages feed), not Luci.
Check that 
* you have installed the openvpn plugin to collectd
* the config file for collectd is created properly by luci-statistics and contains the openvpn item. (symlink ``` /etc/collectd.conf ``` points to ``` /var/etc/collectd.conf ``` that contains the actual config 
* If you manually restart collectd (``` /etc/init.d/collectd restart ```), do you get error messages on console or in system log?

Trunk uses collectd 5.5.0 while CC uses 5.4.2. You might actually start by verifying that it works with the trunk version for you.

runeevjen, 2016-06-18:
Thank you for helping hnyman!

With collectd 5.5.0:
The openvpn plugin was installed, `/etc/collectd.conf` points to `/var/etc/collectd.conf` and contains the openvpn config.
There was an error message related to "RRA" but unfortunately I did not copy it before upgrading collectd.

After this I upgraded collectd to the following (picked from `https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/`:
`root@ruter:~# opkg list_installed | grep collectd
collectd - 5.5.1-6
collectd-mod-conntrack - 5.5.1-6
collectd-mod-interface - 5.5.1-6
collectd-mod-iptables - 5.5.1-6
collectd-mod-iwinfo - 5.5.1-6
collectd-mod-load - 5.5.1-6
collectd-mod-network - 5.5.1-6
collectd-mod-openvpn - 5.5.1-6
collectd-mod-ping - 5.5.1-6
collectd-mod-protocols - 5.5.1-6
collectd-mod-rrdtool - 5.5.1-6`

After rebooting the router, I can access the Luci setup pages but no plugins are showing under Luci Statistics->Graphs.

There are no errors messages when restarting collectd or in the system log after upgrade, but also no graphing.

hnyman, 2016-06-19:
Trunk packages are built for a different kernel, so you can't use those in 15.05 firmwares. (I meant earlier that install a trunk firmware with collect 5.5.1, but I wonder if you tried to install 5.5.1 to a CC15.05 firmware?

runeevjen, 2016-06-27:
Yes, I tried to upgrade the collect packages only on stock CC firmware with no luck..I'm not able to update the router to trunk now - I need to wait for the next release. I will recheck this issue when the next release is available.

cshoredaniel, 2016-12-10:
Closing per last comment.

closed 2016-12-10 by cshoredaniel