openwrt/luci #745: Input is not proper UTF-8, indicate encoding !

umbrellary, 2016-06-20:
- Chrome 51.0.2704.103 (64-bit) －－－－Appear bug
- IE Safari Firefox... －－－－None

Switch OpenWRT to Chinese language will appear the bug as shown below when use Chrome.

[image]
[image]

jow-, 2016-06-20:
This does not look like vanilla LuCI, seems the template got modified. The UTF-8 error can happen if the *.po files are improperly encoded. Can you provide the full set of changes made to LuCI?

umbrellary, 2016-06-21:
The original openwrt15.05 with luci also have this problem.

cshoredaniel, 2017-01-17:
Unfortunately 15.05 for one type of browser only is probably not a critical issue to fix in the stable branch; can you test if it happens with trunk LEDE/OpenWrt and trunk LuCI (or for-17.01)?