openwrt/luci #746: the cbi.js can't support the self define template

pxnj63, 2016-06-20:
for example, I write a resp_value.html to instand of value.html .

`<%+cbi/resp_valueheader%>
  	<% if self.password then %><div class="input-group"> <% end %>
	<input data-update="change"<%=
		attr("id", cbid) ..
		attr("name", cbid) ..
		attr("type", self.password and "password" or "text") ..
		attr("class", self.password and "resp-input-password form-control preview-password" or "resp-input-text form-control beautify") ..
		attr("value", self:cfgvalue(section) or self.default) ..
		ifattr(self.size, "size") ..
		ifattr(self.placeholder, "placeholder") ..
		ifattr(self.readonly, "readonly") ..
		ifattr(self.maxlength, "maxlength") ..
		ifattr(self.datatype, "data-type", self.datatype) ..
		ifattr(self.datatype, "data-optional", self.optional or self.rmempty) ..
		ifattr(self.combobox_manual, "data-manual", self.combobox_manual) ..
		ifattr(#self.keylist > 0, "data-choices", { self.keylist, self.vallist })
	%> />
  	<% if self.password then %></div> <% end %>
<%+cbi/resp_valuefooter%>
`

this issue was existed on the cc until the trunk version

jow-, 2016-06-20:
... and what is the actual problem?

pxnj63, 2016-06-27:
this issue will generate wrong html structure. and let some js does not work. for example the select depend relation.

jow-, 2016-06-27:
Its up to your template to implement those. Can you provide a minimal, complete testcase?

pxnj63, 2016-07-15:
I have to modify the cbi.js , it is a difficult work for me . is there design documents for the cbi.js ?

cshoredaniel, 2016-12-10:
@pxnj63 No such documents.  This is very much a case of reading the source code and figuring out how it works yourself, or with the help of forums.

jow-, 2018-07-03:
Not LuCI upstream related.

closed 2018-07-03 by jow-