openwrt/luci #757: luci-app-wol not appear in the network network menu

zhangguangtong, 2016-07-13:
I installed luci-app-wol, however it didn't appear in network menu.

After compare other lua file the controller. I come with a fix.

Append '.dependent = true' to the two line start with entry.
```
module("luci.controller.wol", package.seeall)

function index()
        entry({"admin", "network", "wol"}, cbi("wol"), _("Wake on LAN"), 90).dependent = true
        entry({"mini", "network", "wol"}, cbi("wol"), _("Wake on LAN"), 90).dependent = true
end

```

jow-, 2016-07-13:
It should work without that change too after `rm /tmp/luci-indexcache` - can you please test?

zhangguangtong, 2016-07-13:
@jow- 

Thanks, I tested in my side. ```rm /tmp/luci-indexcache``` does works.

I am not clear how ```.dependent = true``` works . Would it better if we add this?

jow-, 2016-07-13:
No, its an obsolete property which should not affect things. I'm closing this PR as the original issue was a stale menu cache.

closed 2016-07-13 by jow-