openwrt/luci #760: luci-app-aria2 unable save other methode

hostkit, 2016-07-16:
@kuoruan can you fix this? 
when i want to other method rpc-scret, luci-app-aria2 said this
[image]

kuoruan, 2016-07-17:
OK, I will fix it. Please wait.

kuoruan, 2016-07-24:
Please try the new version, and give me a feed back. thx

hostkit, 2016-07-25:
no authentication unable to save, user and password luci-app-aria2 stuck when click save

kuoruan, 2016-07-26:
Everything works well in my device.

hostkit, 2016-07-27:
[image]

hostkit, 2016-07-27:
can you build for me :) maybe such as aria2 :smile:

kuoruan, 2016-07-28:
https://soft.kuoruan.com/openwrt/ar71xx/packages/luci/luci-app-aria2_git-16.206.66468-6f5c466-1_all.ipk

hostkit, 2016-07-28:
ok work again :+1: 

suggestion: add download schedule option (start download at ... and stop download at ... )
cat it posible? until now im using manual cronjob using curl

https://github.com/ziahamza/webui-aria2/issues/126 
:smile:

closed 2016-07-30 by hostkit