openwrt/luci #765: Internal DSL modem options

Matthias84, 2016-07-27:
Currently users have to set the Annex manually via SSH and uci:

> uci set network.dsl.annex='b'

This is confusing, as all other settings for the router can be done via Luci already, so people will expect it to be there / not needed to be changed manually.
It would be pretty nice, if it would be possible to change the hardware settings directly via luci :+1: 

(This touches softly also #490 )

jow-, 2016-07-27:
What settings besides annex are missing?

Matthias84, 2016-07-28:
AFAIK this depends on the [DSL hardware driver](https://wiki.openwrt.org/doc/hardware/xdsl), I can just refer to [Lantiq](https://wiki.openwrt.org/doc/hardware/soc/soc.lantiq#lantiqxdsl)
> network.dsl=adsl
> network.dsl.firmware='/lib/firmware/adsl.bin'
> network.dsl.annex='b2p'

There seem to be some more options as tone sequences, but I guess you don't need to alter them usually. Some say they had to patch the DSL modem firmware, so this might be useful, too,

cshoredaniel, 2017-01-17:
Calling this an enhancement because it is functionality not present and merely omits the ability to use one feature of the device (even if an important one like DSL on a DSL modem).

mbo2o, 2018-08-11:
In Release 18.06.0 (and possibly earlier) these settings appear to be implemented in Luci , this issue should be closed

[image]