openwrt/luci #769: LuCI Lua error for some wireless config.

w2ak, 2016-08-04:
Here is the error page at `http://192.168.10.1/cgi-bin/luci/admin/network/wireless` :
```
/usr/lib/lua/luci/util.lua:34: bad argument #2 to 'format' (no value)
stack traceback:
	[C]: in function 'format'
	/usr/lib/lua/luci/util.lua:34: in function </usr/lib/lua/luci/util.lua:29>
	/usr/lib/lua/luci/model/network.lua:1597: in function 'shortname'
	/usr/lib/lua/luci/controller/admin/network.lua:77: in function 'v'
	/usr/lib/lua/luci/dispatcher.lua:563: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:220: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

This happened just after changing my `/etc/config/wireless` file :
```
config wifi-iface
        option device 'radio0'
        option network 'wwan'
        option mode 'monitor'
```

jow-, 2016-08-04:
I pushed a series of changes that should fix your problem.

 * 6a11f71 luci-base: cbi.js: handle undefined arguments in formatting
 * bd45344 luci-mod-admin-full: status page formatting fix
 * 123f14b luci-base: fix handling of monitor ifaces
 * 240b8c4 luci-base: improve string format error reporting

closed 2016-08-04 by jow-

w2ak, 2016-08-04:
Thanks :+1: