openwrt/luci #775: sha256sum support for sysupgrade image checksums?

hnyman, 2016-08-13:
Luci calculates and shows the MD5 checksum for the sysupgrade images. However, as the MD5 checksum calculation has been removed from LEDE and only SHA256 is done since ( https://git.lede-project.org/?p=source.git;a=commitdiff;h=83175687c81bee594885d0738cfd488673f5a3fb;hp=27854a0a84cd2db9539b662a1f1fc95ef7409962 ), LEDE users are left without a checksum verification possibility.

I propose that we add SHA256 support to the Luci sysupgrade page by showing both MD5 and SHA256 checksums for the image.

However, that will also require adding "sha256sum" command to busybox, so this is not a pure Luci exercise. For that reason I am not yet committing this.

The below patches 
* add sha256sum command to busybox defaults. Seems to add some 1.5 kB to the binary (ar71xx).
* modify Luci to show both MD5 and SHA256 checksums. (If sha256sum command is not defined, the field content is empty.)

@jow- 
any opinions?

Luci patch (adds the calculation to admin-full):
```
diff --git a/modules/luci-mod-admin-full/luasrc/controller/admin/system.lua b/modules/luci-mod-admin-full/luasrc/controller/admin/system.lua
index fd6dda1..04d503b 100644
--- a/modules/luci-mod-admin-full/luasrc/controller/admin/system.lua
+++ b/modules/luci-mod-admin-full/luasrc/controller/admin/system.lua
@@ -185,6 +185,10 @@ local function image_checksum(image)
 	return (luci.sys.exec("md5sum %q" % image):match("^([^%s]+)"))
 end
 
+local function image_sha256_checksum(image)
+	return (luci.sys.exec("sha256sum %q" % image):match("^([^%s]+)"))
+end
+
 local function supports_sysupgrade()
 	return nixio.fs.access("/lib/upgrade/platform.sh")
 end
@@ -268,6 +272,7 @@ function action_sysupgrade()
 		if image_supported(image_tmp) then
 			luci.template.render("admin_system/upgrade", {
 				checksum = image_checksum(image_tmp),
+				sha256check = image_sha256_checksum(image_tmp),
 				storage  = storage_size(),
 				size     = (fs.stat(image_tmp, "size") or 0),
 				keep     = (not not http.formvalue("keep"))

diff --git a/modules/luci-mod-admin-full/luasrc/view/admin_system/upgrade.htm b/modules/luci-mod-admin-full/luasrc/view/admin_system/upgrade.htm
index 5ca0398..32fd2bd 100644
--- a/modules/luci-mod-admin-full/luasrc/view/admin_system/upgrade.htm
+++ b/modules/luci-mod-admin-full/luasrc/view/admin_system/upgrade.htm
@@ -24,7 +24,8 @@
 
 <fieldset class="cbi-section">
 	<ul>
-		<li><%:Checksum%>: <code><%=checksum%></code></li>
+		<li><%:Checksum MD5%>: <code><%=checksum%></code></li>
+		<li><%:Checksum SHA256%>: <code><%=sha256check%></code></li>
 		<li><%:Size%>: <%
 			local w = require "luci.tools.webadmin"
 			write(w.byte_format(size))

```
Busybox patch:
```
diff --git a/package/utils/busybox/Config-defaults.in b/package/utils/busybox/Config-defaults.in
index 5240616..105a3fc 100644
--- a/package/utils/busybox/Config-defaults.in
+++ b/package/utils/busybox/Config-defaults.in
@@ -705,7 +705,7 @@ config BUSYBOX_DEFAULT_SHA1SUM
 	default n
 config BUSYBOX_DEFAULT_SHA256SUM
 	bool
-	default n
+	default y
 config BUSYBOX_DEFAULT_SHA512SUM
 	bool
 	default n
```

hnyman, 2016-08-19:
@jow- 
I noticed that you committed the busybox sha256 change, so that part has been taken care of.

Is my proposal for Luci good enough to commit in?

jow-, 2016-08-19:
Yep, it is fine - feel free to push and thanks for taking care of it.

closed 2016-08-19 by jow-

hnyman, 2016-08-19:
commited as https://github.com/openwrt/luci/commit/c0af3601fb87e6c695d07f0af1fbec8dcfdb2f71