openwrt/luci #779: Luci - Network switch config not viewable

shogsbro, 2016-08-18:
Built trunk LEDE last night for my HH2B, with full Luci. I copied switch configuration [from here](http://openwrt.ebilan.co.uk/viewtopic.php?f=4&t=49), and then see a Switch menu entry under Network. 

When selected, I get this Lua error:

    /usr/lib/lua/luci/dispatcher.lua:460: Failed to execute cbi dispatcher target for entry '/admin/network/vlan'.
    The called action terminated with an exception:
    /usr/lib/lua/luci/model/cbi/admin_network/vlan.lua:276: attempt to index local 'topo' (a nil value)
    stack traceback:
    	[C]: in function 'assert'
    	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
    	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

Here is relevant parts of `/etc/config/network`:

    config switch 'eth0'
            option name             eth0
            option reset            1
            option enable_vlan      1

    config switch_vlan              # LAN
            option vlan             1
            option device           eth0
            option ports            '1 2 3 5t'

    config switch_vlan              # WAN
            option vlan             2
            option device           eth0
            option ports            '4 5t'

    config switch_port
            option port             4
            option pvid             2

Output from `swconfig dev eth0 show`:

    root@lede:~# swconfig dev eth0 show
    Global attributes:
    	enable_vlan: 1
    Port 0:
    	mib: Port 0 MIB counters
    RxPacket    : 0
    RxByte      : 0
    TxPacket    : 0
    TxByte      : 0
    Collision   : 0
    Error       : 0
    
    	pvid: 0
    	link: port:0 link:down
    Port 1:
    	mib: Port 1 MIB counters
    RxPacket    : 0
    RxByte      : 0
    TxPacket    : 0
    TxByte      : 0
    Collision   : 0
    Error       : 0
    
    	pvid: 1
    	link: port:1 link:down
    Port 2:
    	mib: Port 2 MIB counters
    RxPacket    : 0
    RxByte      : 0
    TxPacket    : 0
    TxByte      : 0
    Collision   : 0
    Error       : 0
    
    	pvid: 1
    	link: port:2 link:down
    Port 3:
    	mib: Port 3 MIB counters
    RxPacket    : 0
    RxByte      : 0
    TxPacket    : 0
    TxByte      : 0
    Collision   : 0
    Error       : 0
    
    	pvid: 1
    	link: port:3 link:down
    Port 4:
    	mib: Port 4 MIB counters
    RxPacket    : 26926
    RxByte      : 3160224
    TxPacket    : 5867
    TxByte      : 644135
    Collision   : 0
    Error       : 0
    
    	pvid: 2
    	link: port:4 link:up speed:100baseT full-duplex txflow rxflow auto
    Port 5:
    	mib: Port 5 MIB counters
    RxPacket    : 5870
    RxByte      : 644725
    TxPacket    : 26929
    TxByte      : 3160504
    Collision   : 0
    Error       : 0
    
    	pvid: 0
    	link: port:5 link:up speed:100baseT full-duplex txflow rxflow auto
    VLAN 1:
    	vid: 1
    	ports: 1 2 3 5t 
    VLAN 2:
    	vid: 2
    	ports: 4 5t 

I have TTL serial port access, should that be relevant.

shogsbro, 2016-08-18:
I followed the code, and see it is looking for switch info in `/etc/board.json`, which looks like this:

    # cat /etc/board.json 
    {
    	"model": {
    		"id": "BTHOMEHUBV2B",
    		"name": "BTHOMEHUBV2B - BT Home Hub 2B"
    	},
    	"led": {
    		"wifi": {
    			"name": "wifi",
    			"type": "trigger",
    			"sysfs": "bthomehubv2b:blue:wireless",
    			"trigger": "phy0tpt"
    		},
    		"internet": {
    			"name": "internet",
    			"type": "netdev",
    			"sysfs": "bthomehubv2b:blue:broadband",
    			"device": "nas0",
    			"mode": "link tx rx"
    		}
    	},
    	"network": {
    		"lan": {
    			"ifname": "eth0",
    			"protocol": "static",
    			"macaddr": "00:21:04:da:4b:98"
    		},
    		"wan": {
    			"ifname": "nas0",
    			"protocol": "pppoe",
    			"macaddr": "00:21:04:da:4b:99"
    		}
    	},
    	"dsl": {
    		"atmbridge": {
    			"vpi": 1,
    			"vci": 32,
    			"encaps": "llc",
    			"payload": "bridged"
    		},
    		"modem": {
    			"type": "adsl",
    			"annex": "a",
    			"firmware": "\/lib\/firmware\/adsl.bin"
    		}
    	}
    }

I'll try to work out what populates this file (and isn't adding the switch entries required).

shogsbro, 2016-08-18:
So `/etc/board.json` should be populated by `/etc/board.d/02_network`, which is missing the switch section for the `BTHOMEHUBV2B`, probably something like this:

    	ucidef_add_switch "switch0" \
		"0:lan:1" "1:lan:2" "2:lan:3" "4:lan:4" "5t@eth0"

shogsbro, 2016-08-18:
It looks like the code protects itself from lacking data in `/etc/board.json`:

                    if swc and topo then                                                                                            

I have not worked out how to output debugging info in LuCI code, but possibly this bug is caused by the defaulting to "eth0" at line 533 in [this changeset](https://github.com/openwrt/luci/commit/55f458326dc77dd35dce2c22df2353b2a0192627).

closed 2016-08-18 by jow-

seeb73, 2023-04-20:
How to add a switch Xiaomi Redmi AC2100 Router (white)?

commit c390a09c (2016-08-18): luci-mod-admin-full: fallback to generic switch topology

Use a generic default topology if we cannot infer switch information from
board.json and display an error message in this case. Should fix #779 and
similar issues.