openwrt/luci #780: Package libiwinfo required! - but it is there!?

n8bs, 2016-08-20:
Router TRENDnet TEW-691GR (TEW-692GR is also affected)
I'm running the LEDE snapshot image 2016.08.20 and have installed luci-ssl.
In the gui I cannot enable wlan.
In Network -> Wireless it says in a yellow box:
**Package libiwinfo required!**
 The _libiwinfo-lua_ package is not installed. You must install this component for working wireless configuration!

But from opkg it looks like it is there allready.
`root@lede:~# opkg install libiwinfo-lua`
`Package libiwinfo-lua
(2016-07-04-96d322d5941c57be58f8e8d5c6e2b28e90ed3598) installed in root is up to date.`
`root@lede:~# opkg install libiwinfo`
`Package libiwinfo (2016-07-04-96d322d5941c57be58f8e8d5c6e2b28e90ed3598) installed in root is up to date.`

I tried reflash an install non-ssl luci. Same problem.

n8bs, 2016-08-22:
I'm on LEDE Reboot r1399 / LuCI Master (git-16.232.29304-c0af360) 
Configured wireless from commandlien without any issues but in luci I get **Package libiwinfo required!**

mkresin, 2016-08-25:
@jow- The error is related to the new shared build and the fact the mipsel_74kc is used by brcm47xx and rt3883. Therefore the wl iwinfo backend will be included for the rt3883 devices without have the wl kmod(?).

Long story short, libiwinfo needs a PKG_FLAGS:=nonshared.

```
root@lede:~# ldd /usr/lib/lua/iwinfo.so
        ldd (0x5558e000)
        libuci.so => /lib/libuci.so (0x773bc000)
        libiwinfo.so => /usr/lib/libiwinfo.so (0x773a0000)
        liblua.so.5.1.5 => /usr/lib/liblua.so.5.1.5 (0x77372000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7734e000)
        libc.so => ldd (0x5558e000)
        libubox.so => /lib/libubox.so (0x77334000)
        libnl-tiny.so => /usr/lib/libnl-tiny.so (0x7731c000)
Error relocating /usr/lib/lua/iwinfo.so: wl_ops: symbol not found

root@lede:~# lua -l iwinfo
lua: error loading module 'iwinfo' from file '/usr/lib/lua/iwinfo.so':
        Error relocating /usr/lib/lua/iwinfo.so: wl_ops: symbol not found
stack traceback:
        [C]: ?
        [C]: ?
        [C]: ?
root@lede:~#
```

jow-, 2016-08-25:
Spot-on, @mkresin :) I pushed a fix to the tree.

closed 2016-08-25 by jow-

n8bs, 2016-08-25:
Confirmed. Working in r1442.