openwrt/luci #784: luci-theme-material:Switch port status is hidden after using this theme.

981213, 2016-08-29:
I think it might be caused by the following function defined at line 202 in htdocs/luci-static/material/js/script.js:
`
    $(".cbi-section-table-titles, .cbi-section-table-descr, .cbi-section-descr").each(function () {
        var that = $(this);
        if (that.text().trim() == ""){
            that.css("display", "none");
        }
    });
`
but I knew little about JavaScript so I don't know how to fix the problem :-(
@LuttyYang I guess this function will hide cbi-section-table-descr if there is no plain text?
But the port status is defined like this:
`
<tr class="cbi-section-table-descr">
<th class="cbi-section-table-cell"><div id='portstatus-switch0'></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-6"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-7"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-0"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-1"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-2"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-3"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-5"></div></th>
<th class="cbi-section-table-cell"><div id="portstatus-switch0-4"></div></th>
</tr>
`
and it will be hidden.

finasfinas, 2016-12-26:
I can confirm this on a TPLink C5 with recent trunk

LGA1150, 2018-06-26:
Confirmed in r7312
Can be fixed by removing the `.cbi-section-table-descr, ` in that line.

jow-, 2018-06-26:
Fixed on latest LuCI master (not 18.06):

[image]

closed 2018-06-26 by jow-