openwrt/luci #785: Add a danish locale for luci-base

rubo77, 2016-08-29:
I think it would be really easy to add a danish locale to the base, just copy the norwegian `base.po` and correct the minor differences.

Would this work, If I just copy it and add the setting like this?

```
NEW=da
BASE=no
cp -a packages/luci/modules/luci-base/po/$BASE/ packages/luci/modules/luci-base/po/$NEW/
sed -i 's/Language: $BASE/Language: $NEW/g' packages/luci/modules/luci-base/po/$NEW/base.po
```

And add this line manually in gluon/packages/luci/luci.mk:

LUCI_LANG.da=Dansk (Danish)

rubo77, 2016-08-29:
we would need a basic danish support, so we could start to implement it in Gluon: gluon.readthedocs.io/en/v2016.1.5/dev/i18n.html

hnyman, 2016-08-29:
It is that simple in principle.

But luci-base is not enough, as the translation is distributed to the various packages. For the minimal core functionality you would additionally need the .po file at least for luci-app-firewall
https://github.com/openwrt/luci/blob/master/applications/luci-app-firewall/po/no/firewall.po

And possibly also luci-app-meshwizard and other apps that you need for Gluon purposes.

If you create a PR for adding the Danish translation, it can be merged in.

hnyman, 2016-08-29:
One additional piece of advice how to add a new language to Luci. It is not really documented anywhere and it is a bit tricky to get "make menuconfig" and/or "make defconfig" to recognise the new language packages, as the language packages are designed to be hidden:

1) Add new LUCI_LANG definition to luci.mk . For example "fi" for Finnish:
```
 LUCI_LANG.fi=Suomi (Finnish)
```

2) Initialise translation for luci-base. Go to the "po" directory and create a dir for the new language.
```
cd modules/luci-base/po
mkdir fi
msginit -o fi/base.po -i templates/base.pot -l fi
```
(It is also possible to copy the .po file from a similar language to speed up the translation, as long as you edit the header info to reflect the correct language. e.g, using Norwegian for Danish might work)

3) Update the packages and the index, and install the packages. This lays the foundation for menuconfig to find the new language and the new luci-i18n-base-fi package. First remove the existing index and its tmp files, then update things.
```
rm feeds/luci.index
rm -rf feeds/luci.tmp
./scripts/feeds update
./scripts/feeds install -a
```

4) Clear the config system's cache. This is the tricky step and can be easily forgotten as it is not that obvious. "make dirclean" takes care of this, but a more lightweight alternative is to just delete the "tmp" directory with "rm -rf tmp" in the buildroot. 
```
rm -rf tmp
```
And now make menuconfig should find the new language.

I struggled a long time myself before I figured out that last step. As long as there are old tmp/info/.packageinfo-feeds_luci_luci-* files, the new language packages will not show up in menuconfig.

rubo77, 2016-08-29:
That sounds great, can you please create a branch with the Norwegian language files as base for the Danish, then we would do just the translations

hnyman, 2016-08-31:
There is no need to create a branch for that here. You can do the editing in your local repo without that. (And the branches here in the main repo are just for the official releases.)

It also serves no purpose to commit the Norwegian .po file here as Danish. You can use that for editing in your local repo, but it does not need to be committed to the main repo.

You should do the necessary editing in your local fork (copy of this repo) and then send the changes as a "pull request" for importing them here. You might start just with luci.mk and base.po, and then continue with other files after those initial ones get imported.

rubo77, 2016-09-01:
Will this be enough to add the danish language to luci?

https://github.com/openwrt/luci/compare/master...rubo77:danish?expand=1

Then we'll start translating the norwegian to danish right away

hnyman, 2016-09-01:
That looks almost ok to me. Just some technical comments and advice:
* In luci.mk, the new "Danish" line should be before "Deutsch", not after.  da<de
* In two of the .po files, you have left the dates & names of the last Norwegian translator. Please remove those or even better, place your own info there..
* Remember that the commit message (the git commit message itself, not the PR) should be "signed off by" properly by you. Read  https://github.com/openwrt/luci/blob/master/CONTRIBUTING.md
* If you commit the editing to your local repo in several separate commits, please squash them together with ``` git rebase -i HEAD~6``` etc. so that there is just one commit, before you create the PR. Some advice in CONTRIBUTING.md.
* It pays off to have a separate local git branch for the work. Then you can easily do the possibly needed editing to the PR (even after sending the PR).

I suggest that after editing the .po files, but before committing them, you run from the Luci repo's root directory the command ```  ./build/i18n-sync.sh ```. That will straighten up e.g. the line wraps of multiline strings, as the line lengths may have changed.

rubo77, 2016-09-05:
It is already quite advanced: now you only have to edit https://github.com/rubo77/luci/blob/danish/modules/luci-base/po/da/base.po and use the comparisn between norwegian and danisch, i created here: https://github.com/rubo77/luci/commit/4eee2f97c76afc03d2149895efa80d547c6b7d66?diff=split

rubo77, 2016-09-06:
I deleted the untranslated norwegian lines and created new empty lines with the template.

Add basic danish translation #790

closed 2016-09-06 by rubo77